dotintent/react-native-ble-plx #384: How to use hands free bluetooth?

AideProjet, 2019-01-04:
Hi, 

We're trying to use this library inside an app made to be used in a car.

We want to listen to audio messages with the car's bluetooth and record the radio at the same time (with the phone mic). 

Problem is, we seem to be unable to record radio and use bluetooth at the same time. 

We believe hands free protocol might be a solution, I couldn't find any information on its use with this library, any idea? 

Thanks!

dariuszseweryn, 2019-01-04:
Why do you think it is an issue with the library?

AideProjet, 2019-01-04:
Hey, thanks for your answer. 

I don't think it's an issue, sorry if I'm in the wrong category, I want to know how to switch from A2DP protocol to HFP protocol with the library, any idea? 

Thanks

dariuszseweryn, 2019-01-04:
This is a BLE library (hence `react-native-ble-plx`) it has no support for A2DP and I have no information about HFP but I doubt it is implemented on BLE

AideProjet, 2019-01-04:
Ok, my bad, any chance you have knowledge of a way to have support A2DP/Hands Free Protocol with React Native? 

Thanks

dariuszseweryn, 2019-01-04:
Unfortunately not

AideProjet, 2019-01-04:
Ok thanks man

closed 2019-01-04 by dariuszseweryn