dotintent/react-native-ble-plx #387: Could not run example

ithieund, 2019-01-06:
I pulled the fresh repository, go to example folder and type react-native android-run and this error display:

Android project not found. Maybe run react-native android first?

I also tried to run npm install and lots of errors display:
[image]

Could you help?

ithieund, 2019-01-12:
@userException ,
It does not work
[image]

dariuszseweryn, 2019-01-12:
The example folder inside the library is not maintained for a long time already. [The official docs](https://github.com/Polidea/react-native-ble-plx#configuration--installation) have links to other projects that use this library though.

ithieund, 2019-02-11:
> The example folder inside the library is not maintained for a long time already. [The official docs](https://github.com/Polidea/react-native-ble-plx#configuration--installation) have links to other projects that use this library though.

Thanks for pointing out.
I think that link should be state obviously in Example section, and the example folder should be remove d to prevent confusion.

Cierpliwy, 2019-04-05:
Example is already removed from the repo. Closing.

closed 2019-04-05 by Cierpliwy