dotintent/react-native-ble-plx #390: BleError: Cannot start scanning operation on certain Android Phones

dested, 2019-01-09:
I am getting the above error on certain android phones when trying to initiate a scan. I am trying to track down exactly what version and model the phones are, but do you have any insight into what would cause this error to be thrown? The call I am making is ```manager.startDeviceScan([serviceId], {allowDuplicates: false}, async (error, device) => {});```

Is it possible that this version of android doesn't support either allowDuplicates false or scanning by serviceId?

Cierpliwy, 2019-01-11:
One note: callback shouldn't be "async", `startDeviceScan` expects just normal function.

Can you specify which Android phones fail and what are their versions? Verbose logs from logcat would be helpful as well.

hsmade, 2019-01-29:
I seem to have the same issue.
HW: Lenovo p2 (eu)
Android: 7.0

Logcat:
`01-29 22:18:15.924 23913 23932 I ReactNativeJS: Running application "App" with appParams: {"rootTag":1}. __DEV__ === false, development-level warning are OFF, performance optimizations are ON
01-29 22:18:15.943 23913 23932 I ReactNativeJS: permission granted
01-29 22:18:15.944 23913 23932 I ReactNativeJS: BLE manager started
01-29 22:18:15.955  2626  2722 D BluetoothManagerService: Message: 20
01-29 22:18:15.955  2626  2722 D BluetoothManagerService: Added callback: android.bluetooth.IBluetoothManagerCallback$Stub$Proxy@2c663d6:true
01-29 22:18:16.010 23913 23960 D BluetoothAdapter: STATE_ON
01-29 22:18:16.015 12479 12491 D BtGatt.GattService: registerClient() - UUID=06c92c0b-e058-47dd-b0b7-3d1277960ccf
01-29 22:18:16.016 12479 12510 E bt_att  : GATT_Register: can't Register GATT client, MAX client reached!
01-29 22:18:16.016 12479 12510 E bt_btif : Register with GATT stack failed.
01-29 22:18:16.016 12479 12510 E bt_att  : GATT_Register: can't Register GATT client, MAX client reached!
01-29 22:18:16.016 12479 12510 E bt_btif : Register with GATT stack failed.
01-29 22:18:16.016 12479 12510 E bt_att  : GATT_Register: can't Register GATT client, MAX client reached!
01-29 22:18:16.016 12479 12510 E bt_btif : Register with GATT stack failed.
01-29 22:18:16.017 12479 12504 D BtGatt.GattService: onClientRegistered() - UUID=06c92c0b-e058-47dd-b0b7-3d1277960ccf, clientIf=0
01-29 22:18:16.018 23913 23925 D BluetoothLeScanner: onClientRegistered() - status=133 clientIf=0 mClientIf=0
01-29 22:18:16.022 23913 23960 D BluetoothAdapter: STATE_ON
01-29 22:18:16.022 23913 23960 D BluetoothLeScanner: could not find callback wrapper
`

Same apk/code does work on a Find 7 with android 8.1

hsmade, 2019-01-30:
When I disable bluetooth, then start the app, wait for a bit and re-enable bluetooth, the scanning works. However, it disconnects after a few seconds. This smells of an Android bug

Cierpliwy, 2019-02-01:
> 01-29 22:18:16.016 12479 12510 E bt_att  : GATT_Register: can't Register GATT client, MAX client reached!

This is generally bad. It means that all applications on Android used up all GATT clients and therefore your app won't be able to do anything.

Please make sure that your application has ACCESS_COARSE_LOCATION enabled (it is a dangerous permission) as well as location enabled (GPS symbol).

hsmade, 2019-02-01:
Hey,

I have both enabled. It works on another phone. I can also get it to work after a reboot, if no other apps started running. I'm guessing this is either a bug in android, or some issue with one of the apps (nrf connect?) that I have installed. Not much to be fixed here in code, I'm afraid. Apart from maybe be a bit more verbose in the error message (why was scanning not started?), as we have that info in the java BleError class.

Cierpliwy, 2019-04-05:
Closing, as this is probably not an issue in library itself.

closed 2019-04-05 by Cierpliwy

VihangaAshinsana, 2019-06-02:
I got the same error(`BleError: Cannot start scanning operation`). The call I'm making is the same method which @dested was trying to call( `this.manager.startDeviceScan`) I did the following

- Restart my phone(Huawei GR5 2017: Android Nougat) and tried to call the method(No other apps were running).
- Tried using a different device (Nokia 8: Android Pie )
How can i sort this out?
Thanks

jbreuer95, 2019-12-02:
I ran into this problem and found some conditions when it occurs and how to prevent it;

1. It mostly seems to happen on cheaper android phones.
2. When you are developing and have fast refresh enabled make sure you properly destroy the old ble manager and start a new one. If you don't, you start multiple ble managers and multiple scans at the same time. This will cause this error for sure. Put a console.log in your scan function to monitor for this. 
3. Dont make your scans very short. I get this sometimes when my scans are 1 or 2 seconds. I now have them at 6 seconds and i dont see this error anymore

Davidson-09, 2022-10-29:
how do I increase scan time ?