dotintent/react-native-ble-plx #392: How do i continuously read data that send by the ble device?

Dhiraj2996, 2019-01-14:


dariuszseweryn, 2019-01-16:
> how do I read that data continously?

You start monitoring the characteristic that sends the data

closed 2019-01-16 by dariuszseweryn

Dhiraj2996, 2019-01-19:
Hey thanks for the reply! So i am looking for notifiable characteristics,but there are more than one available. This is my function to find characteristic to monitor.
Is there a way to find the required characteristic without hard coding uuid (as it may be used for other ble devices)?
` getNotifyServicesAndCharacteristics(device) {
    return new Promise((resolve, reject) => {
      device.services().then(services => {
        const characteristics = []

        services.forEach((service, i) => {
          service.characteristics().then(c => {
            characteristics.push(c)

            if (i === services.length - 1) {
              const temp = characteristics.reduce((acc, current) => {
                return [...acc, ...current]
              }, [])
              const dialog = temp.find(
                characteristic => characteristic.isNotifiable
              )
              if (!dialog) {
                reject('No notifiable characteristic')
              }
              resolve(dialog)
            }
          })
        })
      })
    })
  }`
thanking you in advance!

dariuszseweryn, 2019-02-01:
> Is there a way to find the required characteristic without hard coding uuid (as it may be used for other ble devices)?

You usually identify a characteristic by UUID because you as the user of the library know what you are looking for and how to interpret the received data. Alternatively you can iterate and filter the discovered characteristics in any way you want

I hope this answers your question

yangchange, 2019-05-13:
hey, maybe it is late but you solved your problem?