dotintent/react-native-ble-plx #395: Can detect all devices

omaralanei, 2019-01-21:
When i try to get all the nearest devices, i get only one device and a lot of null values

Cierpliwy, 2019-01-25:
Can you describe your problem better? What platform is it? Example code and detailed logs are needed to help you.

kaanoner3, 2019-03-31:
Hi, I am trying to scan all devices which is near to my phone. I am testing it IPhone X. I have two devices one of them is jbl charge 3, the other one my macbook. bleManager found my macbook properly but it didn't find my jbl.  If only jbl is open for connection, manager is not returning anyting either result or error. Did I miss something ?

Here is my epic for scanning:

[image]

    "react-native": "0.58.4",

Cierpliwy, 2019-04-05:
@kaanoner3 Does this jbl device support LE connections or only Bluetooth Classic (not supported by this lib)?

kaanoner3, 2019-04-05:
@Cierpliwy I think it doesn't. I tested another device it is working fine.

Cierpliwy, 2019-04-19:
In that case I'll close this issue.

closed 2019-04-19 by Cierpliwy