dotintent/react-native-ble-plx #399: connected devices not found!

bhenav, 2019-01-26:
￼￼￼
### console log
[image]

### iphone bluetooth settings
[image]

### code
```javascript
  constructor() {
    this.manager = new BleManager();
  }

async componentDidMount() {
    if (Platform.OS === 'ios') {
      this.hasBluetoothPermission = await Permissions.check('bluetooth');
      if (this.hasBluetoothPermission === 'undetermined') {
        this.hasBluetoothPermission = await Permissions.request('bluetooth');
      }
    }
    this.manager.onStateChange(async (state) => {
      if (state === State.PoweredOn) {
        this.scanAndConnect();
      } else {
        this.manager.stopDeviceScan();
      }
    }, true);
  }

  scanAndConnect() {
    this.manager.startDeviceScan(null, null, async (error, device) => {
      if (!await this.devices.find(item => item.id === device.id)) {
        this.devices.push(device);
        device.onDisconnected((err, d) => {
          console.log('device Disconnected', d);
        });
        this.devices.forEach(async item => {
          console.log(item.id, item.name, item.isConnectable, await item.isConnected());
        });
        let connectedDevices = await this.manager.connectedDevices(null);
        console.log(connectedDevices);
      }
    });
  }
```

bhenav, 2019-01-29:
?

Cierpliwy, 2019-02-04:
Are you sure that your device supports Bluetooth Low Energy apart from Bluetooth classic?

Cierpliwy, 2019-04-05:
@bhenav Any news?

closed 2019-06-11 by stale[bot]