dotintent/react-native-ble-plx #401: [Android] slow reads and writes

stonehippo, 2019-01-31:
I am seeing very slow reads (and somewhat slow writes) between the Android build of our app and the BLE device. For example, reading a characteristic a single byte value is taking upwards of 1200ms, and can take nearly 5 seconds at times. Write times are also very slow.

The same application running on iOS is seeing considerably faster reads and writes, usually less that 1/5 of the response time.

Using the default MTU of 23, and typically not sending very large values. Any ideas about how to improve read and write performance on Android, or what might be causing the issue?

Cierpliwy, 2019-02-01:
> For example, reading a characteristic a single byte value is taking upwards of 1200ms, and can take nearly 5 seconds at times. 

It looks like a problem with connection interval. Please send me your Android logs from Android Studio during your connection establishment and when you are reading characteristic values. Look for something similar:

2019-02-01 16:40:08.431 4033-4154/? D/BtGatt.GattService: onClientConnParamsChanged() - clientIf=8 address=5C:31:3E:C1:1F:XX, interval=6status=0
2019-02-01 16:40:08.431 6605-6631/com.sensortag D/BluetoothGatt: **onClientConnParamsChanged() - Device=5C:31:3E:C1:1F:XX interval=6 status=0**
2019-02-01 16:40:08.439 6605-7238/com.sensortag D/BluetoothGatt: discoverServices() - device: 5C:31:3E:C1:1F:XX
2019-02-01 16:40:08.440 4033-31630/? D/BtGatt.GattService: discoverServices() - address=5C:31:3E:C1:1F:XX, connId=8

In my case messages are exchanged every 6 * 1.25 ms = 7.5ms. This value could be as high as 4s.
You can try to negotiate lower value by calling `requestConnectionPriorityForDevice` with `ConnectionPriority.High`.

stonehippo, 2019-02-02:
Thanks, I came to a similar conclusion. Checked with the firmware dev and it sounds like they’ve got the connection interval set crazy slow (like 500ms min, 1000ms max). Working to get that down into the 15ms min range, which should work better for both iOS and Android.

closed 2019-02-02 by stonehippo