dotintent/react-native-ble-plx #404: Android: generating APK fail with react-native 0.58.3

alfacommunication-alessandro, 2019-02-05:
**Problem**
Command `gradlew assembleRelease` fail

**Step to reproduce**

1. Start a Native Code project (0.58.3) from scratch. 
2. Install react-native-ble-plx
3. Build for release

**Output**
`java.util.concurrent.ExecutionException: com.android.builder.internal.aapt.v2.Aapt2Exception: Android resource linking failed
  Output:  .../AwesomeProject/node_modules/react-native-ble-plx/android/build/intermediates/res/merged/release/values-v28/values-v28.xml:7: error: resource android:attr/dialogCornerRadius not found.`

**Workaround**
Edit node_modules/react-native-ble-plx/android/build.gradle and change:
`compileSdkVersion 28`
`buildToolsVersion "28.0.2"`

crobinson42, 2019-02-05:
```
Manifest merger failed : uses-sdk:minSdkVersion 16 cannot be smaller than version 18 declared in library [:react-native-ble-plx] /Users/willlyrob/projects/react-native-apps/blePLXDemo/node_modules/react-native-ble-plx/android/build/intermediates/merged_manifests/debug/processDebugManifest/merged/AndroidManifest.xml as the library might be using APIs not available in 16
	Suggestion: use a compatible library with a minSdk of at most 16,
		or increase this project's minSdk version to at least 18,
		or use tools:overrideLibrary="com.polidea.reactnativeble" to force usage (may lead to runtime failures)
```

Cierpliwy, 2019-02-06:
@crobinson42 Please set your app's `minSdkVersion` to 18 or above. It's mentioned in README file.

salimsket, 2019-02-13:
@Cierpliwy does not work, even setting minSdkVersion it to 20/23/27 🙁

Cierpliwy, 2019-02-22:
It should work now without a fix from the above post in version 1.0.3. However, it's an OK solution if you have similar problems with other dependencies Please reopen if the bug occurs again.

closed 2019-02-22 by Cierpliwy