dotintent/react-native-ble-plx #405: BleManager not created when tried cancel device connection

jeppeArchidraw, 2019-02-08:
I am getting some error reports concerning cancellation of device connections. The error is

> IllegalStateException: BleManager not created when tried cancel device connection

> com.polidea.reactnativeble.BleModule in cancelDeviceConnection at line 609
java.lang.reflect.Method in invoke
com.facebook.react.bridge.JavaMethodWrapper in invoke at line 372
com.facebook.react.bridge.JavaModuleWrapper in invoke at line 160
com.facebook.react.bridge.queue.NativeRunnable in run
android.os.Handler in handleCallback at line 789
android.os.Handler in dispatchMessage at line 98
com.facebook.react.bridge.queue.MessageQueueThreadHandler in dispatchMessage at line 29
android.os.Looper in loop at line 164
com.facebook.react.bridge.queue.MessageQueueThreadImpl$3 in run at line 192
java.lang.Thread in run at line 764

I cannot quite figure out why this happens, can anyone help?
This happens in a released application for both Android and iOS, but the error is only reported on Android.

Cierpliwy, 2019-02-22:
This can only happen when:
- You didn't construct BleManager properly. (new BleManager())
- You called `cancelDeviceConnection` after `destroy()` call.
- You created new BleManager instance before closing previous one. Multiple instances are not supported yet.

Cierpliwy, 2019-04-05:
@jeppeArchidraw Any news?

closed 2019-06-11 by stale[bot]