dotintent/react-native-ble-plx #407: monitorCharacteristicForDevice does not return any values

renard12, 2019-02-11:
Hi to all so I need du subscribe to updates for battery lever for my device but monitorCharacteristicForDevice is not working I do something like this:
`    this.manager.monitorCharacteristicForDevice(
      this.props.device.id,
      services.BATTERY_SERVICE_UUID,
      services.BATTERY_LEVEL_CHARACTERISTIC_UUID,
      (err, char) => {
        if (err) {
          return this.showModal(
            'ERROR!',
            'There is some problem with connection, device is disconnected',
            {
              buttons: [
                { text: 'Ok', action: this.disconnectWithDevice },
              ],
            }
          );
        }
        console.log(char);
      }, 'monitor_battery');
  };`

dariuszseweryn, 2019-02-11:
Any info on what you are testing with? Or what is available in the logs?

renard12, 2019-02-11:
Sorry, my bad, I`am doing monitor before discoverAllServicesAndCharacteristics

closed 2019-02-12 by dariuszseweryn