dotintent/react-native-ble-plx #408: How to return CBCentralManager property to Use in nordic-dfu

Manish123Pandia, 2019-02-12:
[RNNordicDfu setCentralManagerGetter:^() {
    return [BleManager getCentralManager];
  }];

  // Reset manager delegate since the Nordic DFU lib "steals" control over it
  [RNNordicDfu setOnDFUComplete:^() {
    NSLog(@"onDFUComplete");
    CBCentralManager * manager = [BleManager getCentralManager];
    manager.delegate = [BleManager getInstance];
  }];
  [RNNordicDfu setOnDFUError:^() {
    NSLog(@"onDFUError");
    CBCentralManager * manager = [BleManager getCentralManager];
    manager.delegate = [BleManager getInstance];
  }];

It works in case of react-native- Ble-manager, but in case of react-native-ble-plx it is not working

dariuszseweryn, 2019-02-18:
You can create a separate `CBCentralManager` — from my experience it works just fine

Cierpliwy, 2019-02-22:
Closing as this is suggested solution in case of mixing multiple BLE libraries.

closed 2019-02-22 by Cierpliwy

Manish123Pandia, 2019-02-22:
Hii [dariuszseweryn](/dariuszseweryn) - Thanks for suggestion can you you please provide a little example code so i can fix this

dariuszseweryn, 2019-02-23:
https://github.com/Polidea/react-native-ble-plx/issues/260