dotintent/react-native-ble-plx #417: IOS :-scanning Error - BluetoothLE is in unknown state

Manish123Pandia, 2019-02-19:
Here is my code : 

startDeviceScan(scanTimeout) {
		var me = this;
		// first destroy BLE manager object and re-initialize it 
		this.manager.destroy();
		this.manager = new BleManager();
		this.manager.startDeviceScan(null, null, (error, device) => {
			if (error) {
				// Handle error (scanning will be stopped automatically)
				console.log("scanning Error - " + error.message);
				me.startDeviceScan();
				return;
			}
			if (device)
				me.emit('startScanSuccess', device);
		});
	}

previously it was working fine for both IOS and Android but after using destroy() it is working fine on android but giving the error in case of IOS.

Cierpliwy, 2019-02-19:
Once a BLEManager instance is created, CBCentralManager is instantiated on iOS. For small period of time BLE stack is in Unknown state, after which it transitions to "PoweredOn" or other one. There is a function which allows you to monitor BLE state. Check code snippet [on introduction page](https://polidea.github.io/react-native-ble-plx/).

Closing as this behaviour is expected.

closed 2019-02-19 by Cierpliwy

roysG, 2019-08-28:
I get this message even when i wait 10 minutes!

'error blueee', [BleError: BluetoothLE is in unknown state]
[68047:2469569] 'error blueee', [BleError: BluetoothLE is in unknown state]

NN01010101, 2019-09-21:
@Cierpliwy @roysG, I can confirm, looks like the Unknown state is always the first response, no matter how long the app has been running. Once Unknown State error has been thrown, state changes to Powered On.

ad-boschung, 2020-07-16:
@NN01010101 You're right it's only an issue when the remote debugger is enabled.   
@Cierpliwy I don't understand why it is NOT documented. It would have earn me a lot of crucial time....

danielfigueroaj, 2020-07-16:
@ad-boschung how do you disable the remote debugger? is it something in xcode or react native?

ad-boschung, 2020-07-17:
It's a React Native feature. It's explained [here](https://reactnative.dev/docs/debugging).

SunstroUS, 2020-08-05:
I think remote debugging is off by default. At least on my build, when I shake my device, it is showing ' Debug ' as an option, meaning it hasn't started debugging yet. I have already tried turning that on, then off again to see if it would produce a different result. Not sure what is happening.

EzeRangel, 2020-12-02:
This also happen when you're using functional components and/or Hooks. Even if you disable the remote debug, still shows the Unknown state Error.

JesusTectronic, 2022-12-05:
var state = await bleManager.state()
while (state == State.Unknown) {
  delay(500)
  state = await bleManager.state()
}

priyasamyal, 2024-01-03:
Hi @ManishKumarPandia  Any update on this. As I am still facing the same issue. Already checked the device State as well