dotintent/react-native-ble-plx #418: Can support write Byte data?

nagaokashi, 2019-02-19:
I've just switched from [react-native-ble-manager](https://www.npmjs.com/package/react-native-ble-manager) to this lib. The react-native-ble-manager uses Bytes format to send data to device, but this  react-native-ble-plx uses Base64. 
I really don't want to modify the protocol of device to adapt base64 format. 
So, is there any way to write data in Bytes instead of Base64 for this lib ?

dariuszseweryn, 2019-02-19:
Hello,

Thanks for using the library.

> So, is there any way to write data in Bytes instead of Base64 for this lib ?

Unfortunately there is no possibility to use byte arrays at this point
> I really don't want to modify the protocol of device to adapt base64 format.

Fortunately you do not have to modify your protocol — it will stay the same. You would just need to map `byte array <=> base64` when interfacing with the library. If you have already migrated then you should have entry points where it would be convenient to do.

A question to broaden my understanding:
Why did you switched from `react-native-ble-manager`?

nagaokashi, 2019-02-19:
Thank you for you answer.
> You would just need to map `byte array <=> base64` **when interfacing with the library**.
That means that I also need to convert data on device to Base64 before send it to phone too?
I'll try.

`Why did you switched from `react-native-ble-manager`?`
Your lib was seem very hot on BLE field. Actually, I just want to do a comparison between 2 libs for further development. Although I thought that both of libs are almost same. All depend on my understanding and programming skill. :dancer:

dariuszseweryn, 2019-02-20:
> That means that I also need to convert data on device to Base64 before send it to phone too?
I'll try.

If you reference your peripheral as device then — no — `base64` is used only for safe transmission through the ReactJS bridge to the native part where it is decoded back to a byte array. It works the same way in the opposite direction — e.g. the native part reads a characteristic and encodes byte array from the peripheral to `base64` in order to send it back to JS code side

nagaokashi, 2019-02-20:
> do not have to modify your protocol
> You would just need to map `byte array <=> base64` when interfacing with the library
> 
I understood.
Thank you very much.

closed 2019-02-20 by nagaokashi

fonsogms, 2021-05-28:
Hi! I am facing a similar issue.
And I am having a few questions.
Turns out like the token I want to send is bigger than 20 bytes so I have to send that information in different chunks until I get authenticated by device.
First question: Can I send those byte arrays packages formatted into Base64 and would that still work?
Second question: What happens if the base64 string is bigger than 20 bytes how does this library deal with it.
Third questions: So when I do a write characteristic with a base64 string would that be sent in byte format to the ble device?
PS: Kudos for the library, I am loving it so far

dariuszseweryn, 2021-05-28:
There is a wiki page that addresses using characteristic values — maybe it would answer your questions?

tevonsb, 2021-11-09:
@fonsogms Are you by chancing using INVERS? Shot in the dark, but we are currently implementing it and would love to compare notes.

tevonsb, 2021-11-09:
@dariuszseweryn could you point me in the right direction for reading more about chunking characteristic values? I've read the wiki and documentation and haven't found the section you reference regarding using characteristic values.

fonsogms, 2021-11-09:
Hi @tevonsb yes I am :)
I managed to make it work with the ble-plx library.
Just make sure that you have the right format in bytes first and then convert it into base64 so you can send it to the invers device.
If you still have more questions ping me on discord fonsigms#4290

jonasblumer, 2022-01-20:
@tevonsb @fonsogms I'm also working in this for invers. Would be great to have some back and forth if possible!

tevonsb, 2022-01-20:
@jonasblumer Happily! @fonsogms literally saved me a week of my life (maybe more), hopefully I can return the favor. 

I'll shoot you an email. 

@jonasblumer @fonsogms Do you think it would be an issue just to post code snippets here? I don't mind sharing my code

fonsogms, 2022-01-21:
Hi @jonasblumer! Feel free to contact me on discord fonsigms#4290 so we can share some notes.
@tevonsb I would rather not share code here since this one is a closed issue on github :/.
We could create a discord channel  so we can share notes with other people :)

Killyannoor, 2022-08-09:
@fonsogms Hey, I am also very much struggling to write characteristics to the INVERS Cloudboxx. Could I send you a message on Discord? My Discord is Killyan#5768. Help would be very much appreciated :)

Killyannoor, 2022-08-11:
@fonsogms Helped me out greatly, thank you so much!