dotintent/react-native-ble-plx #423: pod update fail

p4bl1t0, 2019-02-27:
When running `pod update` I'm seeing the following error:

> [!] Unable to determine Swift version for the following pods:
> 
> - `react-native-ble-plx-swift` does not specify a Swift version and none of the targets (`evd`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod.

My `pod --version ` output is: 

> 1.6.1

Suggestions?

p4bl1t0, 2019-03-01:
After setting SWIFT_VERSION I'm getting

> [!] An error occurred while processing the post-install hook of the Podfile.
> 
> undefined method `native_target' for <Pod::PodTarget name=Amplitude-iOS >:Pod::PodTarget

In order projects said that this is a problem with the cocoapod version

p4bl1t0, 2019-03-01:
Finally that solved the problem with `pod update` but now when I try to build the solution from Xcode I'm getting the following linking error:

> Undefined symbols for architecture arm64:
>   "protocol descriptor for Swift.Collection", referenced from:
>       l_got.$SSlMp in libreact-native-ble-plx-swift.a(Zip+Collection.o)
>   "protocol witness table for Swift.Int64 : Swift.BinaryInteger in Swift", referenced from:
>       react_native_ble_plx_swift.(ZipCollectionType in _BCE3332D9AE148ED20637FC8E9C507A8).init(sources: A, resultSelector: ([A.Element.E]) throws -> B) -> react_native_ble_plx_swift.(ZipCollectionType in _BCE3332D9AE148ED20637FC8E9C507A8)<A, B> in libreact-native-ble-plx-swift.a(Zip+Collection.o)
>   "Swift.Array.subscript.nativePinningMutableAddressor : (Swift.Int) -> A", referenced from:
>       react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8).on(_: react_native_ble_plx_swift.Event<A.Element.E>, atIndex: Swift.Int) -> () in libreact-native-ble-plx-swift.a(Zip+Collection.o)
>   "Swift.Array.init(repeating: A, count: Swift.Int) -> [A]", referenced from:
>       react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8).init(parent: react_native_ble_plx_swift.(ZipCollectionType in _BCE3332D9AE148ED20637FC8E9C507A8)<A, B.E>, observer: B, cancel: react_native_ble_plx_swift.Cancelable) -> react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8)<A, B> in libreact-native-ble-plx-swift.a(Zip+Collection.o)
>   "Swift.Array.init() -> [A]", referenced from:
>       react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8).init(parent: react_native_ble_plx_swift.(ZipCollectionType in _BCE3332D9AE148ED20637FC8E9C507A8)<A, B.E>, observer: B, cancel: react_native_ble_plx_swift.Cancelable) -> react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8)<A, B> in libreact-native-ble-plx-swift.a(Zip+Collection.o)
>       react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8).on(_: react_native_ble_plx_swift.Event<A.Element.E>, atIndex: Swift.Int) -> () in libreact-native-ble-plx-swift.a(Zip+Collection.o)
>   "Swift.Array.reserveCapacity(Swift.Int) -> ()", referenced from:
>       react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8).init(parent: react_native_ble_plx_swift.(ZipCollectionType in _BCE3332D9AE148ED20637FC8E9C507A8)<A, B.E>, observer: B, cancel: react_native_ble_plx_swift.Cancelable) -> react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8)<A, B> in libreact-native-ble-plx-swift.a(Zip+Collection.o)
>       react_native_ble_plx_swift.(ZipCollectionTypeSink in _BCE3332D9AE148ED20637FC8E9C507A8).on(_: react_native_ble_plx_swift.Event<A.Element.E>, atIndex: Swift.Int) -> () in libreact-native-ble-plx-swift.a(Zip+Collection.o)
>   "Swift.String.index(before: Swift.String.Index) -> Swift.String.Index", referenced from:
>       (extension in react_native_ble_plx_swift):Swift.String.lastIndexOf(Swift.Character) -> Swift.String.Index? in libreact-native-ble-plx-swift.a(String+Rx.o)
>   "static Swift.Character.== infix(Swift.Character, Swift.Character) -> Swift.Bool", referenced from:
>       (extension in react_native_ble_plx_swift):Swift.String.lastIndexOf(Swift.Character) -> Swift.String.Index? in libreact-native-ble-plx-swift.a(String+Rx.o)
>   "_swift_getFunctionTypeMetadata2", referenced from:
>       closure #1 (A) -> react_native_ble_plx_swift.Disposable in react_native_ble_plx_swift.AnyRecursiveScheduler.schedule(_: A, dueTime: Swift.Double) -> () in libreact-native-ble-plx-swift.a(RecursiveScheduler.o)
>       closure #1 (A) -> react_native_ble_plx_swift.Disposable in react_native_ble_plx_swift.AnyRecursiveScheduler.schedule(A) -> () in libreact-native-ble-plx-swift.a(RecursiveScheduler.o)
>       closure #1 (A) -> react_native_ble_plx_swift.Disposable in react_native_ble_plx_swift.RecursiveImmediateScheduler.schedule(A) -> () in libreact-native-ble-plx-swift.a(RecursiveScheduler.o)
>   "protocol descriptor for Swift.FixedWidthInteger", referenced from:
>       l_got.$Ss17FixedWidthIntegerMp in libreact-native-ble-plx-swift.a(Timer.o)
>   "dispatch thunk of Swift.FixedWidthInteger.addingReportingOverflow(A) -> (partialValue: A, overflow: Swift.Bool)", referenced from:
>       closure #1 (A.E) -> A.E in react_native_ble_plx_swift.(TimerSink in _7E8A1FFA6C07F53EE32882E6F9B60697).run() -> react_native_ble_plx_swift.Disposable in libreact-native-ble-plx-swift.a(Timer.o)
>   "dispatch thunk of Swift._ExpressibleByBuiltinIntegerLiteral.init(_builtinIntegerLiteral: Builtin.Int2048) -> A", referenced from:
>       react_native_ble_plx_swift.(TimerSink in _7E8A1FFA6C07F53EE32882E6F9B60697).run() -> react_native_ble_plx_swift.Disposable in libreact-native-ble-plx-swift.a(Timer.o)
>       closure #1 (A.E) -> A.E in react_native_ble_plx_swift.(TimerSink in _7E8A1FFA6C07F53EE32882E6F9B60697).run() -> react_native_ble_plx_swift.Disposable in libreact-native-ble-plx-swift.a(Timer.o)
>       closure #1 (react_native_ble_plx_swift.(TimerOneOffSink in _7E8A1FFA6C07F53EE32882E6F9B60697)<A>) -> react_native_ble_plx_swift.Disposable in react_native_ble_plx_swift.(TimerOneOffSink in _7E8A1FFA6C07F53EE32882E6F9B60697).run() -> react_native_ble_plx_swift.Disposable in libreact-native-ble-plx-swift.a(Timer.o)
[... cut ...]
>       closure #1 () -> A? in react_native_ble_plx_swift.Queue.makeIterator() -> Swift.AnyIterator<A> in libreact-native-ble-plx-swift.a(Queue.o)
>       react_native_ble_plx_swift.dispatch<A>(react_native_ble_plx_swift.Bag<(react_native_ble_plx_swift.Event<A>) -> ()>, react_native_ble_plx_swift.Event<A>) -> () in libreact-native-ble-plx-swift.a(Bag+Rx.o)
>       react_native_ble_plx_swift.disposeAll(in: react_native_ble_plx_swift.Bag<react_native_ble_plx_swift.Disposable>) -> () in libreact-native-ble-plx-swift.a(Bag+Rx.o)
>

p4bl1t0, 2019-03-02:
Finally I start a new Expo fresh project and I was able to correct link the library following the installation steps.

closed 2019-03-02 by p4bl1t0

anovacap, 2019-03-13:
> After setting SWIFT_VERSION I'm getting
> 
> > [!] An error occurred while processing the post-install hook of the Podfile.
> > undefined method `native_target' for <Pod::PodTarget name=Amplitude-iOS >:Pod::PodTarget
> 
> In order projects said that this is a problem with the cocoapod version

Where did you set SWIFT_VERSION? What file is it in?

p4bl1t0, 2019-03-14:
@anovacap don't follow that path.

Finally the only way thay I found to install this module is to start a fresh project.

I set the SWIFT_VERSION  in the build settings of the xcode project

christianortega123, 2019-04-03:
Hi, I still can't find where you set the SWIFT_VERSION in the build settings of the xcode project. Where exactly is it? Thanks! @p4bl1t0

p4bl1t0, 2019-04-03:
I really don't re-call. Do not mess with the SWIFT_VERSION. Try starting a fresh project.