dotintent/react-native-ble-plx #426: Connecting to "connectedDevices" is cancelled

cndev123, 2019-03-05:
This is on android. I have a service UUID and use it to discover an already connected device with blem.connectedDevices. 
One device is returned. device.isConnected tells me false. When I try to connect with device.connect(), I get "Operation was cancelled".

Addition: I have debug access to the place where the device is connected to at the beginning. I see that when device.connect() is called in my code, this leads to a disconnect of the device in the other place. Is that normal? Is it possible to connect a peripheral to more than one app at the same time. Is there a notion of a local connect supported by this library, like described here https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/BestPracticesForInteractingWithARemotePeripheralDevice/BestPracticesForInteractingWithARemotePeripheralDevice.html in the section "Retrieving a List of Connected Peripherals".

mrigo, 2019-03-20:
Same on iOS...
After some tries I got "Operation was cancelled" and I can't do anything more...

doozMen, 2019-03-27:
same problem but only on android. Will analyse now

Cierpliwy, 2019-04-05:
Can you prepare minimal code snippet to reproduce the issue?

closed 2019-06-11 by stale[bot]

folofse, 2019-07-17:
I have the same error in iOS, after some poking around in the code I found that the on line 245 in RXBluetoothkit/BluetoothManager.swift, cancelPeripheralConnection and dispose gets called if the peripheral state  isConnected != true. But I think this is called to early, and I don't think the isConnected has been update even if it going to succeed to connect.

By just commenting out cancelPeripheralConnection and dispose I manage to get it to work, but I'm not sure exactly what's happening in the code and it might trigger some other problems.

`return Disposables.create { [weak self] in
                guard let strongSelf = self else { return }
                if !peripheral.isConnected {
                    //strongSelf.centralManager.cancelPeripheralConnection(peripheral.peripheral)
                    //disposable.dispose()
                }
            }`

bwindsor, 2020-06-09:
I have seen this issue on iOS.

I believe it happens when there is already a pending connection attempt when you call `connect`.

Calling `device.cancelConnection()` before calling `device.connect()` fixed this for me.

I dug quite deeply into the iOS native code. I think the source of the problem is actually in [BleModule.swift](https://github.com/Polidea/MultiPlatformBleAdapter/blob/d2279e0621b285f2d16382af668ae3981abf0639/iOS/classes/BleModule.swift#L387).

The line `connectingPeripherals.replaceDisposable(deviceId, disposable: connectionDisposable)` is called after the new connection attempt is started. The previous connection attempt is then aborted by `replaceDisposable`. This immediately cancels the new connection attempt. A solution to this would be to add the line `connectingPeripherals.removeDisposable(deviceId)` _before_ `.subscribe` is called on the new `connectionObservable`. This ensures any old connection attempt is disposed of before the new one is started.