dotintent/react-native-ble-plx #430: not working for me

weinixuehao, 2019-03-12:
Fuck! The lib has no example! use this lib not working to me! 
to monitor the device for notification but never receive any notifications

piotrdubiel, 2019-03-12:
Please check [this project](https://github.com/Cierpliwy/SensorTag) for example usage.

jimji1005, 2019-03-13:
Not sure if an ass or just bad eyesight

weinixuehao, 2019-03-16:
@piotrdubiel 
Sorry for my bad attitude.
I found that i installed two library which are react-native-ble-manager and react-native-ble-plx
when i intead of the library with the react-native-ble-manager library.
but i have not yet proved whether they have conflict  because i am buzy and used react-native-ble-manager debug success! 

Sorry for my poor english!

Cierpliwy, 2019-03-22:
You should not mix multiple BLE libraries as it may be problematic. How can I help you otherwise?

mariomeraz, 2019-04-09:
> Please check [this project](https://github.com/Cierpliwy/SensorTag) for example usage.

Hi again, finally install in my Device Android, open the App, check the permission and Log seee start Scan.

After a few minutes it does not detect anything, something I'm doing wrong?

 Mi BT BLE temperature sensor and gps
[image]

[image]

Cierpliwy, 2019-04-19:
@mariomeraz Did you modify the source code of the example app to filter proper devices during a scanning procedure?

Cierpliwy, 2019-04-19:
Look at: https://github.com/Cierpliwy/SensorTag/blob/master/src/Saga.js#L150

closed 2019-06-25 by stale[bot]