dotintent/react-native-ble-plx #433: Question

pkt-dev, 2019-03-15:
## Question

Why would the id when scanning for a device be different between Android and iOS for the same exact ble device? Android shows the id that is listed on the device screen, but iOS shows a GUID.

piotrdubiel, 2019-03-15:
That's because iOS hides MAC address and generates its own id. Unfortunately there's no easy around this. If you can change device's firmware, you can add separate characteristic with constant device identifier.

Cierpliwy, 2019-03-15:
Both MAC address on Android and UUID generated by the iOS share the same trait: they uniquely identify the device. That's why they are under the same property.

pkt-dev, 2019-03-15:
These devices all have the same localName, but different ids. So I have no idea which device is the correct device I'm connecting to if there are multiple in the area. There is manufacturerData, but that doesn't look like it has the mac address in there.

Cierpliwy, 2019-03-15:
The only option to uniquely identify the same device between multiple platforms during the advertising phase is to use `manufacturerData` field. However, it's a firmware developer responsibility to define it and fill with useful data.

pkt-dev, 2019-03-15:
Would this have any effect on writing characteristics to a device? I'm able to connect and read/write on Android. On iOS I'm able to connect but when I try to write it never returns.

Cierpliwy, 2019-03-22:
What do you mean by your last sentence? Asynchronous call to write doesn't complete?

closed 2019-05-28 by stale[bot]