dotintent/react-native-ble-plx #434: Reading data from heart rate sensor

Zeyad-Zaky, 2019-03-20:
I am trying to make an application to read the heart rate from a sensor (mioVelo) and I managed to successfully connect to the sensor but I do not know , how to read the heart rate from the sensor and which methods to use to get the required data.

Cierpliwy, 2019-03-22:
Did you check documentation: https://polidea.github.io/react-native-ble-plx/? If this short intro is not sufficient for you, could you give me some ideas about how we could improve on that?

Zeyad-Zaky, 2019-03-22:
Yes, I have checked it and I tried to check the characterstics of all the services of the sensor, neither of them was the heart rate, they were either info about the device such as the name or undefined values. So is there another way to try to reach the required value?

Cierpliwy, 2019-04-05:
Could you provide more information how services/characteristics structure looks like in form of logs?

jimji1005, 2019-04-12:
if the ble device maker did not release any information on how to connect, how to read data etc, chances are you won't be able to read it. I would imagine most company would be adding some kind of logic to prevent ANYONE from just connecting to device and reading from it.

closed 2019-06-18 by stale[bot]