dotintent/react-native-ble-plx #435: Application hangs during startDeviceScan due to which the user is unable to navigate back or press any button.

kkiran16, 2019-03-21:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed

## Expected Behavior

Please describe the behavior you are expecting

## Current Behavior

What is the current behavior?

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: X.Y.Z
* Platform: Android/iOS.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```js
function example(): string {
  return "Your code sample"
}
```

Cierpliwy, 2019-03-22:
Every operation provided by a library is asynchronous. Nothing should be blocked. Could you create a repo with your example, so that I could test it more easily?

closed 2019-05-28 by stale[bot]

tomdaniel0x01, 2019-11-02:
My React Native UI would also hang while performing a `startDeviceScan`. The cause was having the Debugger enabled. Once I disabled the debugger the UI responded lighting fast while `startDeviceScan` was executing.

webjay, 2023-10-22:
Mine was blocked in dev when I had `{ allowDuplicates: true }`.