dotintent/react-native-ble-plx #436: linker command failed with exit code 1

BeauCarrel, 2019-03-22:
A linker command failed with exit code 1 appears every time I try to compile with the current version (1.0.3) 

### Steps to Reproduce

I have tried with a new and otherwise empty rn project and I have the same result

```
react-native init appname

npm install -save react-native-ble-plx

react-native link react-native-ble-plx
```

Then, when I compile in Xcode, I get a "A linker command failed with exit code 1"

The entire message is huge so I won't paste it all unless you want me to, but here is the end:

```
  "_swift_weakLoadStrong", referenced from:
      closure #1 ([BleClient.RxServiceType]?, Swift.Error?) throws -> BleClient.Observable<[BleClient.Service]> in BleClient.Peripheral.discoverServices([__C.CBUUID]?) -> BleClient.Observable<[BleClient.Service]> in libBleClient.a(Peripheral.o)
      closure #2 () -> () in BleClient.Peripheral.discoverServices([__C.CBUUID]?) -> BleClient.Observable<[BleClient.Service]> in libBleClient.a(Peripheral.o)
      closure #2 (BleClient.RxServiceType, Swift.Error?) throws -> BleClient.Observable<[BleClient.Service]> in BleClient.Peripheral.discoverIncludedServices(_: [__C.CBUUID]?, for: BleClient.Service) -> BleClient.Observable<[BleClient.Service]> in libBleClient.a(Peripheral.o)
      closure #3 () -> () in BleClient.Peripheral.discoverIncludedServices(_: [__C.CBUUID]?, for: BleClient.Service) -> BleClient.Observable<[BleClient.Service]> in libBleClient.a(Peripheral.o)
      closure #3 () -> () in BleClient.Peripheral.discoverCharacteristics(_: [__C.CBUUID]?, for: BleClient.Service) -> BleClient.Observable<[BleClient.Characteristic]> in libBleClient.a(Peripheral.o)
      closure #2 (Swift.Bool) -> BleClient.Characteristic in BleClient.Peripheral.writeValue(_: Foundation.Data, for: BleClient.Characteristic, type: __C.CBCharacteristicWriteType) -> BleClient.Observable<BleClient.Characteristic> in libBleClient.a(Peripheral.o)
      closure #3 () -> BleClient.Observable<BleClient.Characteristic> in BleClient.Peripheral.writeValue(_: Foundation.Data, for: BleClient.Characteristic, type: __C.CBCharacteristicWriteType) -> BleClient.Observable<BleClient.Characteristic> in libBleClient.a(Peripheral.o)
      ...
  "_swift_willThrow", referenced from:
      reabstraction thunk helper from @escaping @callee_guaranteed (@unowned BleClient.BluetoothState) -> (@owned BleClient.RestoredState?, @error @owned Swift.Error) to @escaping @callee_guaranteed (@in_guaranteed BleClient.BluetoothState) -> (@out BleClient.RestoredState?, @error @owned Swift.Error) in libBleClient.a(BleModule.o)
      reabstraction thunk helper from @escaping @callee_guaranteed (@guaranteed BleClient.RestoredState) -> (@owned BleClient.RestoredState?, @error @owned Swift.Error) to @escaping @callee_guaranteed (@in_guaranteed BleClient.RestoredState) -> (@out BleClient.RestoredState?, @error @owned Swift.Error) in libBleClient.a(BleModule.o)
      reabstraction thunk helper from @callee_guaranteed (@unowned Foundation.UUID, @guaranteed BleClient.Peripheral) -> (@error @owned Swift.Error) to @escaping @callee_guaranteed (@in_guaranteed (key: Foundation.UUID, value: BleClient.Peripheral)) -> (@error @owned Swift.Error) in libBleClient.a(BleModule.o)
      reabstraction thunk helper from @callee_guaranteed (@guaranteed BleClient.Characteristic) -> (@error @owned Swift.Error) to @escaping @callee_guaranteed (@in_guaranteed BleClient.Characteristic) -> (@error @owned Swift.Error) in libBleClient.a(BleModule.o)
      reabstraction thunk helper from @callee_guaranteed (@guaranteed BleClient.Service) -> (@error @owned Swift.Error) to @escaping @callee_guaranteed (@in_guaranteed BleClient.Service) -> (@error @owned Swift.Error) in libBleClient.a(BleModule.o)
      reabstraction thunk helper from @callee_guaranteed (@guaranteed BleClient.Peripheral) -> (@error @owned Swift.Error) to @escaping @callee_guaranteed (@in_guaranteed BleClient.Peripheral) -> (@error @owned Swift.Error) in libBleClient.a(BleModule.o)
      reabstraction thunk helper from @callee_guaranteed (@guaranteed Swift.String) -> (@unowned Foundation.UUID?, @error @owned Swift.Error) to @escaping @callee_guaranteed (@in_guaranteed Swift.String) -> (@out Foundation.UUID?, @error @owned Swift.Error) in libBleClient.a(BleModule.o)

ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

My package.json:

```
{
  "name": "BMS_BLE",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest"
  },
  "dependencies": {
    "react": "16.8.3",
    "react-native": "0.59.1",
    "react-native-ble-plx": "^1.0.3"
  },
  "devDependencies": {
    "@babel/core": "^7.4.0",
    "@babel/runtime": "^7.4.2",
    "babel-jest": "^24.5.0",
    "jest": "^24.5.0",
    "metro-react-native-babel-preset": "^0.53.1",
    "react-test-renderer": "16.8.3"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

Xcode version: 10.1

I have also tried in my actual projet, which is sporting react native version 0.57.0 with the same error. I also tried an older version of this library (1.0.1) with the same resulting error.

I'm aware that there is another issue with the same error, but the solution did not help me. I have flowed the install instructions to the letter. Though I don't fully understand step 3: "Add empty Swift file if you don't have at least one" So I have not done it, could someone explain the use of creating an empty swift file?

Thanks in advance for the help guys.

Cierpliwy, 2019-03-22:
Yes, this is an essential step. Your project should have at least one Swift file so that XCode will link Swift symbols as well. Can you clarify what is not clear about it?

BeauCarrel, 2019-03-22:
Ah I see, my apologies, I thought it strange to create an empty file, I thought I must have misunderstood something. But it Was to link swift symbols, I see.
I've created it and I am no longer getting this error, sorry for wasting your time and thanks for the speedy response!

Cierpliwy, 2019-03-22:
Glad I could help :)

closed 2019-03-22 by Cierpliwy