dotintent/react-native-ble-plx #437: Question

CaptainJeff, 2019-03-26:
There is a user that is unable to connect to any of the BLE devices. Looking at the logs it appears as though they don't have location services enabled however they have sent me a screenshot of their permissions and they do. The error message says "Location Services disabled (code 4)"

When I try to disable my location services I get a different error which says "Location Permission missing (code 3)"

What is the difference between Location Services Disabled (Code 4) and Location Permission missing (code 3)

dariuszseweryn, 2019-03-26:
On Android an app can have Location Permissions but the user may turned off their Location Services (available usually under the top bar menu or in settings) — it's often referred as "the GPS switch"

CaptainJeff, 2019-03-26:
@dariuszseweryn 

Oh really? I didn't realize that. Is there a way to enable (or request permission to enable) this from the app?

dariuszseweryn, 2019-03-26:
Not sure. There may be something in Google Services. You may read [this](https://github.com/Polidea/RxAndroidBle/issues/554). There is nothing in this library to help you unfortunately

CaptainJeff, 2019-03-26:
@dariuszseweryn understood. Thanks so much for your quick response

closed 2019-03-26 by dariuszseweryn

bwesen, 2025-04-03:
Got the same issue today, on one of my test phones, I had forgotten it had location services disabled and I couldn't figure out why the app asked for BLE permissions properly but never got any scan results. Checking the log I found the "Location services are disabled" error message.

So I guess we have to make sure to catch that error and pop-up a modal asking the users to enable this, but I agree it's not something for this lib - though it would be very helpful to add this issue to the tutorial because with all the location-permission-asking stuff, it's easy to mix it up with an actual location disabling.