dotintent/react-native-ble-plx #438: Getting null in startDeviceScan

Gu7z, 2019-03-30:
Hey,
I have a code with basically this

`search = () => {`
  `manager.startDeviceScan(null, null, (error, device) => {`
    `console.log(device)`
    `manager.stopDeviceScan()`
  `}`
`}`

`<Button onClick={this.search}></Button>`

When i click 5 times in a row, i got null for some seconds then going back to scan. Any ideas to solve?

dariuszseweryn, 2019-03-30:
Have you checked `error`?

Gu7z, 2019-03-31:
i excluded the code and restarted, now it's working but i want to get more than 1 device at a time
Do you know how can i do it?

dariuszseweryn, 2019-04-01:
Do you know what you do in the `startDeviceScan` callback?

CaptainJeff, 2019-04-01:
Yeah, you are stopping the device scan when you return one device

manager.stopDeviceScan()

Cierpliwy, 2019-04-05:
[Type signature](https://polidea.github.io/react-native-ble-plx/#blemanagerstartdevicescan) clearly suggests that Device can be a `null`, when error occurred during scanning. I'm closing this issue, as this is an expected behaviour.

closed 2019-04-05 by Cierpliwy

santiagomartinez, 2019-12-10:
Sorry to open this, but please consider the following limitation on Android 7+:
https://blog.classycode.com/undocumented-android-7-ble-behavior-changes-d1a9bd87d983

You cannot scan more than 5 times in a 30 seconds window or else the scan will fail silently.