dotintent/react-native-ble-plx #440: Two Disconnect Issues

CaptainJeff, 2019-04-01:
I have two disconnect issues that are getting reported with consistency

1)
First one is 

**Disconnected from C4:BE:84:40:1A:BA with status 8 (GATT_INSUF_AUTHORIZATION)**

This issue is happening with a google Pixel running android 8.0

2)
Second one is 

"Disconnected from 98:7B:F3:B0:D7:9D with status 133 (GATT_ERROR)"

**Context**
I have a bluetooth scale that users weigh in on so the duration of the connect is pretty short becuase after the scale is finished processing it shuts off. I am able to connect with the majority of devices without an issue but see these two issues in the logs pretty frequently. Does anyone have any suggestions?

dariuszseweryn, 2019-04-02:
1. Bond the device (it probably does it automatically — check top bar notifications)
2. Android reports some subset of problems this way (i.e. when the peripheral is no longer advertising)

CaptainJeff, 2019-04-02:
@dariuszseweryn 

Sorry can you clarify a little bit

1) Are you suggesting that I should be bonding the device? It seems like we arent' able to explicitly bond to a device
https://github.com/Polidea/react-native-ble-plx/wiki/Device-Bonding

2) Do you know why the peripheral would no longer be advertising? Any suggestions for how to address this issue?

dariuszseweryn, 2019-04-02:
1. Correct
2. Because it went out of range or it stopped advertising I suppose. You even mentioned that your peripheral shuts off quite quickly. Probably nothing can be done about it. Do you always first scan for the peripheral before you connect?

CaptainJeff, 2019-04-02:
1) Do you have any insight or documentation for this library on how to go about bonding. Again I was under the impression this wasn't possible with react-native-ble-plx
2) I don't believe it goes out of range as the phone is right next to the scale as this is happening (this is only related to a few phone types - motorola and pixel). I do always scan first before connecting - is there anything related to that I should be looking out for?

dariuszseweryn, 2019-04-02:
1. Everything is written in the documentation you linked. Either the peripheral will initiate the bonding process or you would need to make a native module that will start it
2. Sometimes `133` happens. You can retry. All known situations that cause this were already mentioned

closed 2019-06-08 by stale[bot]

jasdeepsingh, 2020-10-16:
I'm having this same issue:

"Disconnected from 98:7B:F3:B0:D7:9D with status 133 (GATT_ERROR)"

No idea, why or what is causing this. Any ideas?