dotintent/react-native-ble-plx #441: Please swift version so we can use thanks!

christianortega123, 2019-04-03:
This is the error I'm getting when trying to install this pod

`[!] Unable to determine Swift version for the following pods:

- `react-native-ble-plx-swift` does not specify a Swift version and none of the targets (`timmy1`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod`

Thanks!!

Cierpliwy, 2019-04-05:
Thanks for the report, I'll check that.

jimji1005, 2019-04-23:
@christianortega123 it should work if you just set your swift_version to 4 on the main project. mine is working even without touching the plx-swift library

closed 2019-06-29 by stale[bot]