dotintent/react-native-ble-plx #442: monitorCharacteristicForService value is fragmented

jimji1005, 2019-04-03:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Please describe the behavior you are expecting

monitorCharacteristicForService should receive updates from device completely and not break them into 2 fragments.

## Current Behavior

What is the current behavior?

while using monitorCharacteristicForService to monitor UART protocol sent from device to app, sometimes the app receives an update but library seems to think its 2 fragments.

device sends 20 bytes converted to hex then to base 64
app converts the base 64 back to hex

but when app receives the update, it gets broken into 2 fragments.

Instead of receiving
`QjJBWQE6AACwAQABAgMEBQYHCAk=`
`"42", "32", "41", "59", "01", "3A", "00", "00", "B0", "01", "00", "01", "02", "03", "04", "05", "06", "07", "08", "09"`

The interprets it as 
`QjJB`
`"42", "32", "41", "00", "00", "00", "00", "00", "00", "00", "00", "00", "00", "00", "00", "00", "00", "00", "00", "00"`

and
`WQE6AACwAQABAgMEBQYHCAk=`
`"59", "01", "3A", "00", "00", "B0", "01", "00", "01", "02", "03", "04", "05", "06", "07", "08", "09", "00", "00", "00"`

### Steps to Reproduce &  Context

Testing ON DEVICE, IOS, with proprietary BLE device.

Since device is an inhouse prototype that I can't share, I can only share a few code blocks pertaining to the UART monitor and the parsing

this is the subscription
```
subscribeToUART = (id: string) => {
      // subscribe to UART char

      const device = this.findDeviceInState(id);

      if (!device) {
        return false;
      }

      this.uartSubscription = device.monitorCharacteristicForService(
        UART_UUID,
        RX_UUID,
        this.onUARTSubscriptionUpdate);
    }
```

this is the onUpdate
```
onUARTSubscriptionUpdate = async (error: any, characteristics: any) => {

      if (error) {
        console.log(error);
      } else if (characteristics) {
        console.log('onUARTSubscriptionUpdate', characteristics);

        this.handleMsg(characteristics.deviceID, characteristics.value);

      }

    }
```

this is the parser
```
handleMsg = async (deviceID: string, msg: string) => {
     const bytes = stringToBytes(Base64.atob(msg));
      const hexArray = this.buildPayloadWithPadding(base64toHEX(msg), 20);
      console.log('hexArray', hexArray);
}
```

jimji1005, 2019-04-03:
FYI fragment issue happens kind of random, no way to predict when will it happen or under what conditions. Have not been able to find any pattern to this. Xcode logs does not show any errors

jimji1005, 2019-04-03:
From xcode logs, i did observe logs overlapping

```
[RxBLEKit|DEBG|20:10:26.251]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) writeValue(
[RxBLEKit|DEBG|20:10:26.324]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didWriteValueFor(for:Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10803415584),
[RxBLEKit|DEBG|20:10:36.438]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10803415488),
[RxBLEKit|DEBG|20:10:36.523]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Opt[RxBLEKit|DEBG|20:10:36.655]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didWriteValueFor(for:Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10803415584),
[RxBLEKit|DEBG|20:10:46.625]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10803415488),
[RxBLEKit|DEBG|20:10:46.763]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D8[RxBLEKit|DEBG|20:10:46.842]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didWriteValueFor(for:Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10803415584),
[RxBLEKit|DEBG|20:10:56.774]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10803415488),
[RxBLEKit|DEBG|20:10:56.777]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10803415488),
[RxBLEKit|DEBG|20:10:56.802]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040[RxBLEKit|DEBG|20:10:56.846]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didWriteValueFor(for:Characteristic(uuid: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, id: 10803415584),
```

jimji1005, 2019-04-03:
1 more set of logs from xcode
```
[RxBLEKit|DEBG|20:49:03.946]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779484512),
value: Optional("42324159019b0000"),
error: nil)
[RxBLEKit|DEBG|20:49:03.982]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779484512),
value: Optional("b5493cf9252f9a9b9c9d9e9f"),
error: nil)
```

And this is from the device 
```BTTX: 

42 32 41 59 01 9B 00 00 B5 49 3C F9 25 2F 9A 9B 9C 9D 9E 9F ```

For some reason, ble stack received that TX from above and it got fragmented into 2

jimji1005, 2019-04-03:
Some more interesting results, i intercepted this message here
```[RxBLEKit|DEBG|21:11:52.316]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779492000),
value: Optional("42"),
error: nil)
[RxBLEKit|DEBG|21:11:52.353]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779492000),
value: Optional("324159016b0000b57b1ca4b809babbbcbdbebf"),
error: nil)```

Which if you combine the fragment, you get `42324159016b0000b57b1ca4b809babbbcbdbebf`

However, on RN, i got base64 msg of `Qg==` and `MkFZAWsAALV7HKS4Cbq7vL2+vw==`

When you add those 2 together, you get `Qg==MkFZAWsAALV7HKS4Cbq7vL2+vw==` which in hex becomes 
`"42", "10", "40", "32", "41", "59", "01", "6B", "00", "00", "B5", "7B", "1C", "A4", "B8", "09", "BA", "BB", "BC", "BD"`

It seems that when BLE stack received the msg its correct but when it passed it to RN it added extra characters at the back

Cierpliwy, 2019-04-04:
You can't concatenate and then decode Base64 strings. You will get the wrong results due to padding, which is added in the end.

Cierpliwy, 2019-04-05:
Additionally, based on the following log:
```
[RxBLEKit|DEBG|20:49:03.946]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779484512),
value: Optional("42324159019b0000"),
error: nil)
[RxBLEKit|DEBG|20:49:03.982]: Peripheral(uuid: FCCDB1D5-6FEF-F07B-2E6D-9D269040D849, name: Optional("BMMPR")) didUpdateValueFor(for:Characteristic(uuid: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E, id: 10779484512),
value: Optional("b5493cf9252f9a9b9c9d9e9f"),
error: nil)
```
I think that it is your device, who fragments messages, which are sent to the iOS app. To be absolutely sure, you should sniff packets coming over the radio.

jimji1005, 2019-04-05:
@Cierpliwy 

Thanks for the response, I did find the concat issue and started to concat as bytearray instead, that solved the parsing.

And I also loaded the app on android and had SIMILAR results of fragmented msgs, kind of determined the issue MUST be with the device, not with PLX.

Thanks for bouncing the ideas hehe

closed 2019-04-05 by jimji1005