dotintent/react-native-ble-plx #443: Cannot connect to device using Service UUIDs

nau88xj, 2019-04-05:
## Prerequisites

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior
BleManager.startDeviceScan should emit callback when serviceUUIDs are present on peripheral.

## Current Behavior

BleManager.startDeviceScan does not emit callback when serviceUUIDs  are present.

### Steps to Reproduce

1. Use array of one or more strings to filter peripherals scanned by service UUID.
2. Callback is not fired
3. Set service UUIDs to null
4. Callback is fired for all peripherals

### Context

Current environment:
```js
{
  "react": "16.8.3",
  "react-native": "0.59.3",
  "react-native-ble-plx": "1.0.3"
}
```

The code below demonstrates the problem:
```js
const bleManager = new BleManager();
bleManager.onStateChange(state => {
    if (state !== "PoweredOn") return
    bleManager.startDeviceScan(
        ["053d03fd-00b0-4331-a337-f49f59777484"], // change this value to null to and all peripherals are called
        null,
        (error, scannedDevice) => {
            if (error) console.error(error);
            bleManager.connectToDevice(scannedDevice.id).then(connectedDevice => {
                return connectedDevice.discoverAllServicesAndCharacteristics()
            }).then(connectedDevice2 => {
                return connectedDevice2.services()
            }).then(services => {
                console.log(services.map(x=>x.uuid));
            })
        }
    )
})
```

Cierpliwy, 2019-04-05:
Is `053d03fd-00b0-4331-a337-f49f59777484` present in advertisement data? You can check `scannedDevice.serviceUUIDs` field to check if this UUID is visible. If not, then that's expected behavior.

xkovacikm, 2019-04-07:
I am unfortunately having a same problem.
Some fields are actually missing in connectToDevice promise

    this.manager.startDeviceScan( null, null, (error, device) =>{
        if( error ){ console.log( error ); }

        if( device.id = "24:0A:C4:32:31:32" ){
                console.log( device ); //  serviceUUIDs, localName, rssi, txPowerLevel, are present
                this.manager.stopDeviceScan();
                
                this.manager.connectToDevice( device.id, { autoconnect: true } ).then( deviceConnected => {
                     return deviceConnected.discoverAllServicesAndCharacteristics();
                }).then( deviceConnected2 => {
                    console.log( deviceConnected2 ); // none of fields above are present now
                });
            }
            
        });`

xkovacikm, 2019-04-07:
console.log no.1
id: "24:0A:C4:32:31:32"
isConnectable: null
localName: "Long name works now"
manufacturerData: null
mtu: 23
name: "Long name works now"
overflowServiceUUIDs: null
rssi: -59
serviceData: null
serviceUUIDs: ["4fafc201-1fb5-459e-8fcc-c5c9c331914b"]
solicitedServiceUUIDs: null
txPowerLevel: 3

console.log no.2
id: "24:0A:C4:32:31:32"
isConnectable: null
localName: null
manufacturerData: null
mtu: 23
name: "Long name works now"
overflowServiceUUIDs: null
rssi: null
serviceData: null
serviceUUIDs: null
solicitedServiceUUIDs: null
txPowerLevel: null

nau88xj, 2019-04-09:
> Is `053d03fd-00b0-4331-a337-f49f59777484` present in advertisement data? You can check `scannedDevice.serviceUUIDs` field to check if this UUID is visible. If not, then that's expected behavior.

The serviceUUIDs is null for this and all other devices within range.  I thought that was only populated after scanning for services.  Since that is null, I guess the next best option is to match the peripheral name during a scan of all devices, then discover services and make sure they match?

dariuszseweryn, 2019-04-09:
If there is no other unique data in the advertisement then yes — name is your best shot

closed 2019-04-10 by nau88xj