dotintent/react-native-ble-plx #446: Android's monitorCharacteristicForDevice not working while on iOS is ok

hoffmannjan, 2019-04-08:
- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
         [This seems to be similar issue](https://github.com/Polidea/react-native-ble-plx/issues/152)

I'm using this code and calling `connectToDevice()` & `subscribeToDevice()` & `writeDataToDevice()` in this order using proper formated data, and on iOS I receive data chunks that comes as notifications. As I said, on iOS It works like a charm, on Android I've never received any data as notification.

```import { BleManager } from "react-native-ble-plx";
import { Buffer } from "buffer";
import { bind } from "decko";
import EventEmitter from "EventEmitter";

import { BLE_DATA_RECEIVED } from "../constants";

const SERVICE_UUID = "000000ff-0000-1000-8000-00805f9b34fb";
const CHARACTERISTIC_UUID = "0000ff01-0000-1000-8000-00805f9b34fb";
const CHARACTERISTIC_TRANSACTION_ID = "MONITOR_TEMPO_CHARACTERISTIC";

export default class BleApi {
  constructor() {
    this.manager = new BleManager();
    this.eventEmitter = new EventEmitter();

    this.manager.setLogLevel("Verbose");
  }

  @bind
  scanDevices(callback) {
    if (this.scanning) return;
    this.scanning = true;
    this.manager.startDeviceScan(null, null, callback);
  }

  stopScanning() {
    if (!this.scanning) return;
    this.scanning = false;
    this.manager.stopDeviceScan();
  }

  connectToDevice({ deviceID }) {
    return this.manager.connectToDevice(deviceID).then(device => {
      this.stopScanning();
      return device.discoverAllServicesAndCharacteristics().then(() => {
        // console.log("Discovered all services and characteristics of device: ", device.name);
        return { deviceID, connected: true, discovered: true };
      });
    });
  }

  disconnectDevice({ deviceID }) {
    this.manager.cancelTransaction(CHARACTERISTIC_TRANSACTION_ID);
    return this.manager.cancelDeviceConnection(deviceID);
  }

  subscribeToDevice({ deviceID }) {
    this.manager.monitorCharacteristicForDevice(
      deviceID,
      SERVICE_UUID,
      CHARACTERISTIC_UUID,
      (error, characteristic) => {
        if (error) {
          console.error("Error at receiving data from device", error);
          return;
        } else {
          const response = Buffer.from(characteristic.value, "base64");
          const isLastChunk = response[response.length - 1] === 0;

          const clearResponse = response.slice(3);

          if (isLastChunk) {
[... 94 lines cut ...]
04-08 19:07:57.323 16172-17129/com.myapp.consumermvp D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ff01-0000-1000-8000-00805f9b34fb enable: true
04-08 19:07:57.349 16172-17129/com.myapp.consumermvp D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(14600020)
04-08 19:07:57.351 16172-17213/com.myapp.consumermvp D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(14600020)
04-08 19:07:57.370 16172-16194/com.myapp.consumermvp D/BluetoothGatt: onConnectionUpdated() - Device=34:26:06:62:D0:6A interval=30 latency=0 timeout=400 status=0
04-08 19:07:57.408 16172-16194/com.myapp.consumermvp D/RxBle#BluetoothGatt$1: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
04-08 19:07:57.413 16172-17213/com.myapp.consumermvp D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(14600020) in 63 ms
04-08 19:07:58.317 16172-17128/com.myapp.consumermvp I/ReactNativeJS: SUBSCIREBBED
04-08 19:07:58.372 16172-17129/com.myapp.consumermvp D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(53394107)
04-08 19:07:58.374 16172-17213/com.myapp.consumermvp D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(53394107)
04-08 19:07:58.400 16172-16194/com.myapp.consumermvp D/RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=0000ff01-0000-1000-8000-00805f9b34fb status=0
04-08 19:07:58.407 16172-17214/com.myapp.consumermvp V/RxBle#BleModule: Write to Characteristic(uuid: 0000ff01-0000-1000-8000-00805f9b34fb, id: 8, value: 7B22636F6D6D616E64223A227365745F6D6F6465222C226D6F6465223A2269646C65227D00)
04-08 19:07:58.415 16172-17213/com.myapp.consumermvp D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(53394107) in 39 ms
04-08 19:08:02.143 16172-16788/com.myapp.consumermvp W/System: A resource failed to call response.body().close(). 
    A resource failed to call response.body().close(). 
04-08 19:08:33.697 16172-17249/com.myapp.consumermvp V/FA: Recording user engagement, ms: 151615
04-08 19:08:33.733 16172-17249/com.myapp.consumermvp V/FA: Connecting to remote service
04-08 19:08:33.765 16172-16172/com.myapp.consumermvp D/ScopedContext: WARNING: getPackageName called on ScopedContext
04-08 19:08:33.766 16172-17249/com.myapp.consumermvp V/FA: Activity paused, time: 691974559
04-08 19:08:33.779 16172-16172/com.myapp.consumermvp D/ScopedContext: WARNING: getPackageName called on ScopedContext
04-08 19:08:33.785 16172-16631/com.myapp.consumermvp D/ScopedContext: WARNING: getPackageName called on ScopedContext
04-08 19:08:33.861 16172-16629/com.myapp.consumermvp D/ScopedContext: WARNING: getPackageName called on ScopedContext
04-08 19:08:34.211 16172-17249/com.myapp.consumermvp V/FA: Connection attempt already in progress
04-08 19:08:34.212 16172-17249/com.myapp.consumermvp D/FA: Connected to remote service
04-08 19:08:34.237 16172-17249/com.myapp.consumermvp V/FA: Processing queued up service tasks: 2
```

dariuszseweryn, 2019-04-08:
As you can see in the logs, there are no notifications received from the peripheral you are connected to.

I think that the problem lays in the data you write to the characteristic: `7B22636F6D6D616E64223A227365745F6D6F6465222C226D6F6465223A2269646C65227D00`. It is far too large for the default MTU (23, 20 bytes usable) so the data you write is simply truncated to `7B22636F6D6D616E64223A227365745F6D6F6465` unless your peripheral supports so called [`Queued Write Request`](http://dev.ti.com/tirex/content/simplelink_cc2640r2_sdk_1_40_00_45/docs/blestack/ble_user_guide/html/ble-stack-3.x/gatt.html#queued-writes) which is rarely seen.

iOS by default negotiates higher MTU (its a system dependent behaviour). Yet on Android nothing like this happens. You do not even request any [MTU change](https://github.com/Polidea/react-native-ble-plx/wiki/MTU-Negotiation).

Try using higher MTU value.

hoffmannjan, 2019-04-09:
Yeah, It works, big big thanks!

Great idea to have It as default behavior in the library as you proposed in issue.

Take care and thanks again!

closed 2019-04-09 by hoffmannjan