dotintent/react-native-ble-plx #447: [Android] Negotiate heighest possible MTU by default

dariuszseweryn, 2019-04-08:
## Current Behavior

iOS negotiates MTU by default to the highest value. Android does not unless explicitly instructed,

## New behavior

Both systems negotiate MTU to the highest value by default.

### Acceptance criteria

Please provide detailed steps to accomplish proposed feature.

1. Connect to a peripheral that supports MTU > 23
2. Write to this peripheral's characteristic data that is longer than 20 bytes
3. Verify if the whole data was transmitted

### Breaking changes

None. Only behaviour changes. Old connection options / MTU negotiation API may be deprecated.

This could fix confusion and mitigate issues like #446

Cierpliwy, 2019-04-19:
@dariuszseweryn I guess that you wrote steps to verify implementation (they are not needed to accomplish proposed feature)?

dariuszseweryn, 2019-04-19:
> I guess that you wrote steps to verify implementation

Correct
> they are not needed to accomplish proposed feature

Correct

dariuszseweryn, 2019-04-19:
I have run some tests on devices with Android 5 and on some (e.g. Micromax Canvas) the MTU seems to timeout or return a `GATT_ERROR` if called against a peripheral that supports max MTU=23 (SensorTag)

For this case:
- it should be possible to turn automatic MTU negotiation off (default on) 
- have a good error description/logging/user feedback
- carry on with the connection if MTU negotiation would return an error

I am currently brainstorming. I can work on this feature once settled

dariuszseweryn, 2019-06-18:
Not stale

da1z, 2022-06-16:
Is there a way to get current highest possible MTU for device(mobile phone) to explicitly pass it to device.connect options?

dariuszseweryn, 2022-06-22:
For iOS devices it is well known but for Android it is not possible afaik. You can always pass the maximum value though

abursztynowska, 2023-10-06:
I'm excited to inform you that we've released a new version of `react-native-ble-plx`. Many of the concerns and bugs have been addressed in this update.

If you encounter any issues or have further suggestions, please don't hesitate to open a new issue.

closed 2023-11-02 by abursztynowska