dotintent/react-native-ble-plx #452: Everytime I try to connect to a device, I get an error for device is disconnected. Why is this happening?

JHANAK-DIDWANIA, 2019-04-15:
`scanAndConnect(){
            this.manager.startDeviceScan(null, null, (error, device) => {
            if(error){
                console.log('Error occurred while scanning');
                return;
            }

            if(device.id === MAC ){
                this.manager.stopDeviceScan();

                this.manager.connectToDevice(MAC, {autoConnect:false})
                .then((device) => {
                    this.info("Discovering services and characteristics")
                    console.log('device');
                    return device.services(device.id);
                })
                .then((service) => {
                    this.info("Listening... ")
                    console.log(services)
                })
                .catch((error) => {
                        console.log(error.message);
                        device.cancelConnection();
                        this.scanAndConnect();
                });
            }
        })`

Cierpliwy, 2019-04-19:
Please provide detailed information about an error which occurs and attach Android logs as well.

TailNest, 2019-06-17:
> ```
>            scanAndConnect(){
>                 this.manager.startDeviceScan(null, null, (error, device) => {
>                 if(error){
>                     console.log('Error occurred while [...]

I might help elaborate because I'm having the same problem. I'm using the same code to connect to my bluetooth device, every time, after few seconds from the `connectToDevice` function I get the error "Device [MAC_ADDRESS] was disconnected".

I can't see any other log (not "Discovering services and characteristics", not "Listening...")...the code goes directly in the `catch` clause of the function and print the error. Here's the stackoverflow question: https://stackoverflow.com/questions/56632350/react-native-bluetooth-device-keep-disconnecting

theshohidul, 2019-07-13:
Same problem here, device doesn't connect in android.

ultimate5289, 2019-07-17:
Same problem here too

ndrishinski, 2019-07-23:
Same here also.

closed 2019-09-28 by stale[bot]

pcrozer, 2019-11-08:
Did anyone figure this out - I'm new to this and am experiencing the same thing - any time I try to connect with above code I get error device was disconnected.  Any help would be appreciated thanks!

gpawlik, 2020-04-15:
The problem still exists, any ideas?