dotintent/react-native-ble-plx #454: Failed removal of a registered listener

dgruntz, 2019-04-18:
We are registering a listener on a particular characteristic of a BLE device using

        this.subscription = device.monitorCharacteristicForService(
                serviceUUID,
                characteristicsUUID,
                (error, characteristic) => { ... },
                '1');

and this works fine. We then turn of the BLE device and turn it on again. Since the connection is lost we let the user press a disconnect button which executes

        device.cancelConnection()
        manager.cancelTransaction('1')
        subscription.remove()

But after reconnecting to the device and registering the listener again, the old listener still seems to be registered, i.e. each notification is delivered twice (once over the old and once over the new listener).

How can we assert that the old listener really gets removed?

Cierpliwy, 2019-04-19:
Transaction IDs in the numeric form are used internally [by a library](https://polidea.github.io/react-native-ble-plx/#transactionid). Please test with different type of a transaction ID and let me know if it helps.

dgruntz, 2019-04-23:
We changed the transaction identifier from `1` to an arbitrary string (e.g. `xxx`) and the problem still occurs, i.e. if the BLE device is turned off and if we reconnect with that device (after having cancelled the connection) then the old listener is still registered.

If we cancel the connection, then connect to another device, cancel that connection and connect to the first device, it seems to work fine (i.e. the notifications are only delivered once).

dgruntz, 2019-04-23:
When `device.cancelConnection()` is invoked, the result is `BleError: Device 00:07:80:72:AD:CF is not connected` in case that the device has been turned off. This might explain why the listener will not be removed. We also changed the order of the method invocations in our disconnect method to

    manager.cancelTransaction('1');
    subscription.remove();
    device.cancelConnection()';

but this did not help either.

dgruntz, 2019-04-24:
We also tried to invoke

    manager.cancelTransaction('1');

after reconnection (and also with a different transaction-id which is not numeric), but this did not help either.

dgruntz, 2019-04-24:
We program around this problem, i.e. we ignore all invocations on stale listeners and react only on the last one registered in a particular device.

closed 2019-06-30 by stale[bot]

george-eucare, 2024-03-22:
same issue