dotintent/react-native-ble-plx #455: [iOS] add () to forward @autoclosure parameter

petrbela, 2019-04-21:
## Setup problem

Getting the following error when compiling.

```
/app/node_modules/react-native-ble-plx/ios/RxBluetoothKit/Logging.swift:90:38: error: add () to forward @autoclosure parameter
        log(with: .verbose, message: message)
                                     ^~~~~~~
                                            ()
/app/node_modules/react-native-ble-plx/ios/RxBluetoothKit/Logging.swift:94:36: error: add () to forward @autoclosure parameter
        log(with: .debug, message: message)
                                   ^~~~~~~
                                          ()
/app/node_modules/react-native-ble-plx/ios/RxBluetoothKit/Logging.swift:98:35: error: add () to forward @autoclosure parameter
        log(with: .info, message: message)
                                  ^~~~~~~
                                         ()
/app/node_modules/react-native-ble-plx/ios/RxBluetoothKit/Logging.swift:102:38: error: add () to forward @autoclosure parameter
        log(with: .warning, message: message)
                                     ^~~~~~~
                                            ()
/app/node_modules/react-native-ble-plx/ios/RxBluetoothKit/Logging.swift:106:36: error: add () to forward @autoclosure parameter
        log(with: .error, message: message)
```

### Context

* Library version: 1.0.3
* Platform: iOS
* Xcode: 10.2.1

petrbela, 2019-04-21:
The problem is Xcode now uses Swift 5.0 by default, which requires the `@autoclosure` tag. The proper solution is to migrate RxBluetoothKit to 5.0 (I think it's already underway in the source repo, just hasn't been ported to this package which has a separate copy of RxBluetoothKit).

For now, it can be fixed by specifying Swift 4.2 as the compiler for this package.

```
post_install do |installer|
  installer.pods_project.targets.each do |target|
    if target.name == 'react-native-ble-plx-swift'
      target.build_configurations.each do |config|
        config.build_settings['SWIFT_VERSION'] = '4.2'
      end
    end
  end
end
```

stepanstepan, 2019-05-10:
in my case app compiles with following code:

```
if target.pod_name == 'react-native-ble-plx-swift'
  target.native_target.build_configurations.each do |config|
    config.build_settings['SWIFT_VERSION'] = '4.2'
  end
end

```

closed 2019-07-16 by stale[bot]

njdullea, 2019-10-17:
In case anyone else ends up here, open your ios folder, open the Podfile, add the code provided by petrbela (thank you) to the bottom, then in a terminal switch to the ios folder and run the command 'pod install'

amirhosein5858, 2019-11-07:
hi 
where we have to use this?: 
`
post_install do |installer|
  installer.pods_project.targets.each do |target|
    if target.name == 'react-native-ble-plx-swift'
      target.build_configurations.each do |config|
        config.build_settings['SWIFT_VERSION'] = '4.2'
      end
    end
  end
end
`

njdullea, 2019-11-07:
@amirhosein5858 

Open your ios folder, find the 'Podfile' and add it to the bottom.