dotintent/react-native-ble-plx #457: Linker Error finding Swift Support Libraries

paddlefish, 2019-05-01:
## Prerequisites

Start with a react-native app (not ejected Expo)

## Expected Behavior

Build and run

## Current Behavior

Linker error:
ld: warning: Could not find auto-linked library 'swiftFoundation'
ld: warning: Could not find auto-linked library 'swiftDarwin'
ld: warning: Could not find auto-linked library 'swiftCoreFoundation'
ld: warning: Could not find auto-linked library 'swiftCore'
ld: warning: Could not find auto-linked library 'swiftCoreGraphics'
ld: warning: Could not find auto-linked library 'swiftObjectiveC'
ld: warning: Could not find auto-linked library 'swiftDispatch'
ld: warning: Could not find auto-linked library 'swiftSwiftOnoneSupport'
Undefined symbols for architecture arm64:

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Install react-native-ble-plx using `yarn add`
2. `react-native link`
3. Build-and-Run in Xcode

### Context

The fix is to add a single empty swift file to your main app.  This will trigger Xcode to link in Swift and create a bridging header.

We should update the README to include this simple step

paddlefish, 2019-05-01:
Oh my, I'm sorry, the README already says that and I skimmed over it.  So sorry!

(feels embarrased)

closed 2019-05-01 by paddlefish

myckhel, 2019-09-19:
Following this solution worked for me

https://stackoverflow.com/a/56187043/9241085

lsirdevan, 2019-09-19:
@myckhel That solution worked for me as well. Thanks!

mzruiz, 2020-10-09:
@paddlefish The fix you suggested worked when I experience the issue integrating [opentok-react-native](https://github.com/opentok/opentok-react-native)