dotintent/react-native-ble-plx #458: cant send data more than 128 bytes

amirhosein5858, 2019-05-01:
- I am running the latest version = yes
- I checked the documentation and found no answer = yes
- I checked to make sure that this issue has not already been filed = yes

hi,i want to send an 232 byte string to a rasberry pi ,project is react native(0.59.5) and created by react-native init.
in android every thing is fine but in ios the sent data is not complete.

* Library version: 1.0.3
* Platform: iOS.
* Platform logs : XCODE

```js
this.manager = new BleManager(); //in constractor
send_order(value){
        this.manager.writeCharacteristicWithoutResponseForDevice(
            this.props.item.id,
            res.values.Service_uuid,
            res.values.Char_uuid,
            encode(value)
        )
    }

```

JanErikFoss, 2019-05-07:
Did you set the MTU?

device.connect({ requestMTU: 256 })

amirhosein5858, 2019-05-08:
yes i do,in last try in ios : device.connect({ requestMTU: 256 }) but received mtu is always 185,data that more than 185 byte does not sent at all

closed 2019-07-14 by stale[bot]

MedRaid, 2020-04-24:
same issue i cannot send long bytes 
https://github.com/Polidea/react-native-ble-plx/issues/653