dotintent/react-native-ble-plx #461: Bug Report: "BleError: Device is not authorized to use BluetoothLE"

PhuongNguyenETIT, 2019-05-08:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed

## Expected Behavior

Please describe the behavior you are expecting

## Current Behavior

What is the current behavior?

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. ...

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: X.Y.Z
* Platform: Android/iOS.
* Platform logs (logcat/XCode):
* JS logs:
* Contents of the `package.json` file:
* Formatted code sample or link to a repository:

```js
function example(): string {
  return "Your code sample"
}
```

KrishnanBharath, 2019-05-24:
Check if you have the location permissions turned on for the app.. It should work if it is

ravelab, 2019-07-22:
I have the same issue, here is my permissions in AndroidManifest.xml:

    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
    <uses-sdk android:minSdkVersion="18" />

Thanks

rogerroelofs, 2019-07-26:
I'm having the same problem.  Running RN 0.60.4

PhuongNguyenETIT, 2019-07-28:
thanks you very much. @KrishnanBharath @ravelab @rogerroelofs

closed 2019-07-28 by PhuongNguyenETIT

rmartinez-paack, 2020-04-16:
Hi, I'm having the same issue...

T-101, 2020-04-17:
Having the same issue

drenghel, 2020-05-05:
Me too, 
added it in the three AndroidManifest files (out of despair)

It's weird the more I test, the more it feels like my changes concerning permission are barely read by the toolset (eg, I can remove all of them, aapt still tell me there are some, I can get the name string wrong still works..., but attributes spelled wrong will crash the builder)

hugows, 2021-06-23:
My case is that AndroidManifest.xml is asking for ACCESS_FINE_LOCATION but the dynamic code that requests user permission was still referencing ACCESS_COARSE_LOCATION. Changing to this makes it work:

```
  if (Platform.OS === "android" && Platform.Version >= 23) {
    yield ilog("Scanning: Checking permissions...");

    const enabled = yield call(PermissionsAndroid.check, PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);
    if (!enabled) {
      yield ilog("Scanning: Permissions disabled, showing...");
      const granted = yield call(PermissionsAndroid.request, PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);
      if (granted !== PermissionsAndroid.RESULTS.GRANTED) {
        yield ilog("Scanning: Permissions not granted, aborting...");
        return;
      }
    }
  }
```

luhaimin, 2022-01-24:
> Check if you have the location permissions turned on for the app.. It should work if it is
Android version less than 12 can run, nice