dotintent/react-native-ble-plx #462: iOS not recognizing service, Android does

AdamGold, 2019-05-12:
## Question

I was wondering if anyone else stumbled upon the problem that everything works fine in Android - I managed to connect to the peripheral and send data to it, while in iOS the connection is fine but the same service and characteristics UIDs aren't being recognized at all? I'm not sure it has anything to do with the package itself, maybe it's even firmware related.

Cierpliwy, 2019-05-24:
It's strange. In most cases I have seen this issue other way around. Make sure you call `discoverServicesAndCharacteristics` and try to send me iOS logs from XCode.

closed 2019-07-30 by stale[bot]

AdamGold, 2019-07-30:
For future reference - The characistics and service UUIDs were different on iOS. Had to check with LightBlue.