dotintent/react-native-ble-plx #464: Question about the connection of device

keyz23, 2019-05-16:
Hi, i'm trying to use this library for my application. 
I follow the guide on your site (https://www.polidea.com/blog/ReactNative_and_Bluetooth_to_An_Other_level/ )
I tried entering the device.name || device.id of my ble device but when I start the app I get the following error:
"ERROR: Service 000F0F1-0000-1000-8000-00805 for device ? not found"
Now i don't understand if it a problem of service (but it shouldn't be because the service number is right. )
And I don't understand if it's the connection to the device that doesn't work.
Thank you.

jinqiangg, 2019-05-21:
Hi, I am facing similar issue. Does it have anything to do with Bluetooth pairing password/ pin?

keyz23, 2019-05-21:
I don't think, I have no password / pin. I managed to make the connection but I can't get the services because it gives me that error

trflagg, 2019-05-21:
I am having a similar problem. After running `discoverAllServicesAndCharacteristicsForDevice` I can write to a characteristic, but when I try to read services or characteristics for the connected device I am getting the error reported here.

dariuszseweryn, 2019-05-21:
Have you checked what services/characteristics are available?

trflagg, 2019-05-21:
Yes and I was getting an empty Array.

For me, I wonder if I might be using an old instance of the device.  Looks like certain calls take a device ID but instantiate a new device model. I was holding the device in props and passing the id to BleManager.

I fixed this for `connectToDevice` but looks like I need it for `discoverAllServicesAndCharacteristicsForDevice` too. Do I need to get the new device back from all of these calls? Are there certain ones to look out for?

trflagg, 2019-05-21:
That was my problem. I was not using the updated device returned by `bleManager.discoverAllServicesAndCharacteristicsForDevice()`. Storing the returned device and using that `device.id` for a later characteristic query fixed this error for me.

dariuszseweryn, 2019-05-21:
The only thing that can be persisted between creations of the client is the `deviceId`. After each connection to a BLE peripheral one _must_ perform a service discovery before using characteristics

trflagg, 2019-05-21:
Oh okay, sounds like saving the device id and access everything via `bleManager` is the best way to go then.  Thanks!

keyz23, 2019-05-22:
@dariuszseweryn I follow the guide in the polidea site but i receive always the error. 
You can see the code I used in the first discussion. I don't understand why it continues to give me the error. In theory the services / characteristics are right. Only he still can't find them. What can it be?

dariuszseweryn, 2019-05-22:
I still think that the library is correct. There is little chance that the service and characteristic have the exact same UUID.

keyz23, 2019-05-22:
I'm sorry I didn't understand. Do you think there are problems in the code?
Could it be that the services and features cannot be written but only read?

dariuszseweryn, 2019-05-22:
Compare your `serviceUUID()`, `notifyUUID()`, `writeUUID()`. They are all the same. There is _little_ chance that `serviceUUID` and `notifyUUID`/`writeUUID` should be the same. Try listing all of the services that were discovered and all characteristics that are contained in their respective services. You will see for yourself

keyz23, 2019-05-22:
Ok I'm new in this world, so I'm sorry if I don't understand well and I'll ask you "stupid" questions.
In essence, I have a uuid for the characteristics of Movement (Accelerometer, Gyroscope).
My problem is that my uuid as you can see in the code does not allow me to add 

+ "0" + || + "1" + || + "2" +
in serviceUUID(num), notifyUUID(num) e writeUUID(num)
as well as in the code of the polidea site

dariuszseweryn, 2019-05-22:
Try listing the services _after_ the service discovery is performed. It could go something along these lines:
```javascript
async function printAllServices(device: Device) {
  const services: [Service] = await device.services();
  for (let i = 0; i < services.length; i++) {
    const service: Service = services[i];
    const characteristics: [Characteristic] = await service.characteristics();
    for (let j = 0; j < characteristics.length; j++) {
      const char: Characteristic = characteristics[j];
      console.log(`Characteristic UUID: ${char.uuid}, Service UUID: ${char.serviceUUID}`);
    }
  }
}
```
And show the output

keyz23, 2019-05-22:
I have insert your code there: 
`scanAndConnect(){
      this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
          /* nel caso di un errore nella scansione */
          return
        }
        /* la variabile model si deve recuperare  */
        let model = '0030';
        if (device.name == this.model_dx(model)  ) {
          this.manager.stopDeviceScan();
          device.connect()
            .then((device) => {
              this.info("Discovering services and characteristics ")
              return device.discoverAllServicesAndCharacteristics()
            })
            .then((device) => {
              this.info("Discovering")
              return this.printAllServices(device)
            })

            .catch((error) => {
              this.error(error.message)
            });
        }

      });
    }
    /* Chiusura Scansione e Connessione */

    async  printAllServices(device: Device) {
      const services: [Service] = await device.services();
      for (let i = 0; i < services.length; i++) {
      const service: Service = services[i];
      const characteristics: [Characteristic] = await service.characteristics();
      for (let j = 0; j < characteristics.length; j++) {
      const char: Characteristic = characteristics[j];
      console.log(`Characteristic UUID: ${char.uuid}, Service UUID: ${char.serviceUUID}`);
    }
    }
  }`

But i received error about: Device, Service, Characteristic

dariuszseweryn, 2019-05-22:
These are `flow` annotations. Feel free to remove them `: type`

keyz23, 2019-05-23:
Hi, these are all the UUID that i found withc the console.log:
`05-23 10:15:56.817 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a00-0000-1000-8000-00805f9b34fb, Service UUID: 00001800-0000-1000-8000-00805f9b34fb
05-23 10:15:56.817 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a01-0000-1000-8000-00805f9b34fb, Service UUID: 00001800-0000-1000-8000-00805f9b34fb
05-23 10:15:56.817 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a04-0000-1000-8000-00805f9b34fb, Service UUID: 00001800-0000-1000-8000-00805f9b34fb
05-23 10:15:56.830 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a05-0000-1000-8000-00805f9b34fb, Service UUID: 00001801-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f1-0000-1000-8000-00805f9b34fb, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f2-0000-1000-8000-00805f9b34fb, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f3-0000-1000-8000-00805f9b34fb, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f4-0000-1000-8000-00805f9b34fb, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f5-0001-0008-0000-0805f9b34fb0, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f6-0001-0008-0000-0805f9b34fb0, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f7-0001-0008-0000-0805f9b34fb0, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f8-0001-0008-0000-0805f9b34fb0, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f9-0001-0008-0000-0805f9b34fb0, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0fa-0001-0008-0000-0805f9b34fb0, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0fb-0000-1000-8000-00805f9b34fb, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0fc-0000-1000-8000-00805f9b34fb, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.858 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0fd-0001-0008-0000-0805f9b34fb0, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
05-23 10:15:56.862 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a19-0000-1000-8000-00805f9b34fb, Service UUID: 0000180f-0000-1000-8000-00805f9b34fb
05-23 10:15:56.873 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a29-0000-1000-8000-00805f9b34fb, Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
05-23 10:15:56.873 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a27-0000-1000-8000-00805f9b34fb, Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
05-23 10:15:56.873 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a26-0000-1000-8000-00805f9b34fb, Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
05-23 10:15:56.873 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a28-0000-1000-8000-00805f9b34fb, Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
05-23 10:15:56.873 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a23-0000-1000-8000-00805f9b34fb, Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
05-23 10:15:56.873 21635 21682 I ReactNativeJS: Characteristic UUID: 00002a24-0000-1000-8000-00805f9b34fb, Service UUID: 0000180a-0000-1000-8000-00805f9b34fb`

dariuszseweryn, 2019-05-23:
So does anything look suspicious to you?

keyz23, 2019-05-24:
Maybe that many of the services are the same?

dariuszseweryn, 2019-05-24:
Different to what?

keyz23, 2019-05-24:
If I wanted to use the example code shown on your site, it would be quite complicated (I think) to use it.
For example regarding:

> Characteristic UUID: 00002a19-0000-1000-8000-00805f9b34fb,
> Service UUID: 0000180f-attractions1000-8000-00805f9b34fb
```

In 
this.sensors = {
// i should use 180 for the service
180: "name"
}
```

but the number services( 0000180f) and the characterist(00002a19) number are complety different.

Did I understand how to use it?

dariuszseweryn, 2019-05-24:
So they are indeed different. Remember that samples are just that — samples — to show the idea. But we are getting too far from the original problem:
> "ERROR: Service 0000F0F1-0000-1000-8000-00805F9B34FB for device ? not found"
> Now i don't understand how to do with the the notification (the service number is right. )

and 
```
05-23 10:15:56.857 21635 21682 I ReactNativeJS: Characteristic UUID: 0000f0f1-0000-1000-8000-00805f9b34fb, Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb
```
and
```
    serviceUUID(num) {
      return this.prefixUUID + num + this.suffixUUID
    }

    notifyUUID(num) {
      return this.prefixUUID + num + this.suffixUUID
    }
```

keyz23, 2019-05-24:
Ok let's considered this one: 

> Characteristic UUID: 0000f0f4-0000-1000-8000-00805f9b34fb
> Service UUID: 0000f0f0-0000-1000-8000-00805f9b34fb

```
   this.prefixUUID = "0000"
    this.suffixUUID = "-0000-1000-8000-00805f9b34fb"
    this.sensors = {
      f0f: "Example"
    }

    }

  serviceUUID(num) {
      return this.prefixUUID + num + "0" + this.suffixUUID
    }

  notifyUUID(num) {
      return this.prefixUUID + num + "4" + this.suffixUUID
    }
```

This is the only method?? The problem is that i can't add in this.sensors other sensors.

dariuszseweryn, 2019-05-28:
The most straightforward solution is:
```
    this.prefixUUID = "0000FOF"
    this.suffixUUID = "-0000-1000-8000-00805f9b34fb"
    this.sensors = {
      //...
      4: "Barometer"
      //...
    }

    serviceUUID(num) {
      return this.prefixUUID + "0" + this.suffixUUID // it is always the same for all listed this.sensors
    }

    notifyUUID(num) {
      return this.prefixUUID + num + this.suffixUUID
    }
```

keyz23, 2019-05-28:
Ok I don't want to bore you more xD 
I had thought of using a situation you proposed but there is a problem:

Basically this is all my code: 
```
export default class Main extends Component {
  constructor() {
    super();
    this.manager = new BleManager()
    this.state = {info: "", values: {}}
    this.prefixUUID = "0000F0F"
    this.suffixUUID = "-0000-1000-8000-00805f9b34fb"
    this.suffixUUIDs = "-0001-0008-0000-0805f9b34fb0"
    this.sensors = {
      5: "Acc+Gyro+Mag"
    }

    }

  serviceUUID(num) {
      return  this.prefixUUID + "0" +  this.suffixUUID
    }

  notifyUUID(num) {
      return this.prefixUUID + num +  this.suffixUUIDs
    }

  
  
  info(message) {
      this.setState({info: message})
    }

  error(message) {
      this.setState({info: "ERROR: " + message})
    }

  updateValue(key, value) {
      this.setState({values: {...this.state.values, [key]: value}})
    }

  componentWillMount(){
    const subscription = this.manager.onStateChange((state) => {
      if (state === 'PoweredOn') {
        this.scanAndConnect();
        subscription.remove();
      }
    }, true);
  }
  async requestPermission() {
    try {
      const granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);

      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        this.setState({permissionStatus:'granted'});
      }else if(granted === PermissionsAndroid.RESULTS.DENIED) {
        this.setState({permissionStatus:'denied'});
      }else{
       
      }
    } catch (err) {
      console.error(err)
    }
  }

    scanAndConnect(){
      this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
         return
        }
        ....... // Connection to device // ......
                 this.manager.stopDeviceScan();
          device.connect()
            .then((device) => {
              this.info("Discovering services and characteristics ")
              return device.discoverAllServicesAndCharacteristics()
            })
      
           .then((device) => {
             this.info("SetupNotification")
             return this.setupNotifications(device)
           })

            .catch((error) => {
              this.error(error.message)
            });
        }

      });
    }
      
    async setupNotifications(device) {
        for (const id in this.sensors) {
          
          const service = this.serviceUUID(id)

          const characteristicN = this.notifyUUID(id)
         
          device.monitorCharacteristicForService(service, characteristicN, (error, characteristic) => {
            if (error) {
              this.error(error.message)
              return
            }
            this.updateValue(characteristic.uuid, characteristic.value)
          })
        }
      }

    render() {
       return (
         <View>
           <Text>{this.state.info}</Text>
           {Object.keys(this.sensors).map((key) => {
          return <Text key={key}>
                   {this.sensors[key] + ": " + (this.state.values[this.notifyUUID(key)] || "-")}
                 </Text>
        })}
         </View>
       )
     }

}
```

Using your advice, when I run the app it blocks at 

>  this.info("SetupNotification")

dariuszseweryn, 2019-05-28:
¯\_(ツ)_/¯ It is a trivial programming issue. There are plenty of possible solutions which I expect you can handle at this point
```javascript
this.sensors = {
  // ...
  "0000f0f4-0000-1000-8000-00805f9b34fb": "Barometer",
  "0000f0f5-0001-0008-0000-0805f9b34fb0": "Gyroscope", // looks like an awful typo in the UUID
  // ...
}

serviceUUID() {
  return "0000f0f0-0000-1000-8000-00805f9b34fb"
}
```
Excuse me but this is not the best place to look for help in debugging your app.  You could try using https://codereview.stackexchange.com/ or https://stackoverflow.com I suppose. 

The issue you opened is not an issue with the library. The library clearly stated that a UUID you passed is not available. I have shown how you can easily list the available characteristics UUIDs and UUIDs of their respective services. From this point in it all about how you will pass them to the library to use

keyz23, 2019-05-28:
Ok thank you for your help. I ask you the last question and then we can close this topic :) 
It's possible to connect to two device simultanely concurrently?? 
In my case i have two devices(one right and one left)

dariuszseweryn, 2019-05-28:
https://github.com/Polidea/react-native-ble-plx/wiki/Device-Connecting

> It is possible to have a few connections established at the same time.

keyz23, 2019-05-28:
Ok so it's possible. Thanks!

dariuszseweryn, 2019-05-28:
I guess I can close now

closed 2019-05-28 by dariuszseweryn