dotintent/react-native-ble-plx #466: Can I use This library in React Native Windows app

dharmendrasingh88, 2019-05-21:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel.

## Question

Please describe the question.

Cierpliwy, 2019-05-24:
Currently, this library doesn't support Windows and there are no plans for it right now.

closed 2019-07-30 by stale[bot]

noshaid-ali, 2025-03-03:
Any update on it? Is there any support available for Windows desktop or still not?

noshaid-ali, 2025-03-06:
@dharmendrasingh88 How did you connect BLE devices in react native windows desktop app?