dotintent/react-native-ble-plx #468: BleError: Characteristic X notify change failed for device Y and service Z

packetstracer, 2019-05-21:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [X] I am running the latest version
- [X] I checked the documentation and found no answer
- [X] I checked to make sure that this issue has not already been filed. I've read some similar issues that say that issue has been solved on version 1.0.2, but seems to haven't been solved.

[https://github.com/Polidea/react-native-ble-plx/issues/229](url)
[https://github.com/Polidea/react-native-ble-plx/issues/264](url)
[https://github.com/Polidea/RxAndroidBle/issues/202](url)

## Expected Behavior

subscribe to service's characteristic via characteristic.monitor() method and receive values broadcasted by the peripheral device when callback function is executed

## Current Behavior

subscribe to service's characteristic via characteristic.monitor() method and receive the following error when callback function is executed

`BleError: Characteristic d8b2c95b-b317-47ca-ac02-ccfd3d85a666 notify change failed for device 80293EA3-F380-A86F-B84F-DFC1A5B23502 and service ef66ccad-f4a8-4c51-b025-bc2a2c9fdaa6`

### Steps to Reproduce

1. Connect to device
```js      
        device.connect()
          .then((device) => {
            this.setInfo("Discovering services and characteristics")
            return device.discoverAllServicesAndCharacteristics()
          })
          .then((device) => {
            this.setInfo("Setting notifications")
            return this.setupNotifications(device)
          })
          .then(() => {
            this.setInfo("Listening...")
          }, (error) => {
            this.setError(error.message)
          })
```

2. Access the characteristic and monitor it
```js      
     let voltageCharacteristic = null

      snifferService = services.filter(service => service.uuid === SERVICE_SNIFFER_UUID)[0]

      snifferService.characteristics()
        .then(characteristics => {
          voltageCharacteristic = characteristics.filter(c => c.uuid === SNIFFER_VOLTAGE_UUID)[0]
          voltageCharacteristic.read().then(res => {
            voltageCharacteristic.monitor((error, c) => {
              // getting the error here (voltageCharacteristic has notifiable attribute set to true)
            })
          })
        })
        .catch(error => console.log(error))
```

### Context

BLE Peripheral Device: ESP32 DEVKIT v1

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 1.0.3
* Platform: iOS.
* Platform logs (logcat/XCode):
* JS logs:
`"BleError: Characteristic d8b2c95b-b317-47ca-ac02-ccfd3d85a666 notify change failed for device 80293EA3-F380-A86F-B84F-DFC1A5B23502 and service ef66ccad-f4a8-4c51-b025-bc2a2c9fdaa6

at parseBleError (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:112674:18)

at monitorListener (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:113739:50)

at RCTDeviceEventEmitter.emit (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:6131:37)

at MessageQueue.__callFunction (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:2587:44)

at blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:2344:17

at MessageQueue.__guard (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:2541:13)

at MessageQueue.callFunctionReturnFlushedQueue (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:2343:14)

at http://10.80.148.183:8081/debugger-ui/debuggerWorker.js:80:58"`

* Contents of the `package.json` file:
`{
  "name": "voltageApp",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest",
    "android": "react-native run-android",
    "ios": "react-native run-ios"
  },
  "dependencies": {
    "react": "16.8.3",
    "react-native": "0.59.8",
    "react-native-ble-plx": "^1.0.3",
    "react-native-gesture-handler": "^1.2.1",
    "react-navigation": "^3.9.1"
  },
  "devDependencies": {
    "@babel/core": "^7.4.4",
    "@babel/runtime": "^7.4.4",
    "babel-jest": "^24.8.0",
    "jest": "^24.8.0",
    "metro-react-native-babel-preset": "^0.54.0",
    "react-test-renderer": "16.8.3"
  },
  "jest": {
    "preset": "react-native"
  }
}`
* Formatted code sample or link to a repository:

```js
  scanAndConnect () {
    this.manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        // Handle error (scanning will be stopped automatically)
        this.setError(error)
        return false
      }

      // Check if it is a device you are looking for based on advertisement data or other criteria.
      if (device.name === 'XXX' || device.localName === 'Sniffer') {
        this.device = device
        this.manager.stopDeviceScan()

        device.connect()
          .then((device) => {
            return device.discoverAllServicesAndCharacteristics()
          })
          .then((device) => {
            return this.setupNotifications(device)
          })
          .then(() => {
            this.setInfo("Listening...")
          }, (error) => {
            this.setError(error.message)
          })
      }
    })
  }

  async setupNotifications(device) {
    let snifferService = null

    device.services().then(services => {
      let voltageCharacteristic = null

      snifferService = services.filter(service => service.uuid === SERVICE_SNIFFER_UUID)[0]
      snifferService.characteristics()
        .then(characteristics => {
         // voltageCharacteristic is retrieved correctly and data is also seems correct
          voltageCharacteristic = characteristics.filter(c => c.uuid === SNIFFER_VOLTAGE_UUID)[0]
          voltageCharacteristic.monitor((error, c) => {
           // RECEIVED THE ERROR HERE (voltageCharacteristic.notifiable === true)
          })
        })
        .catch(error => console.log(error))
    })
  }
```

packetstracer, 2019-05-22:
Tried with version 1.0.2 and same error occurs

dariuszseweryn, 2019-05-22:
Hello, have you considered adding native logs?

packetstracer, 2019-05-22:
don't know how to, but going to investigate to provide those

packetstracer, 2019-05-22:
@dariuszseweryn are these the native logs you mean? I've filtered ("BLE" term) and copied them from Xcode

`
[RxBLEKit|DEBG|13:46:34.566]: CentralManager(10775823040) didUpdateState(state: poweredOn)
[RxBLEKit|DEBG|13:46:34.633]: CentralManager(10775823040) scanForPeripherals(
[RxBLEKit|VERB|13:46:34.650]: CentralManager(10775823040) didDiscover(peripheral: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")),
[RxBLEKit|VERB|13:46:34.653]: CentralManager(10775823040) didDiscover(peripheral: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")),
[RxBLEKit|VERB|13:46:34.658]: CentralManager(10775823040) didDiscover(peripheral: Peripheral(uuid: 5B26683F-2F2E-3561-EAB7-96B738DF3F5E, name: nil),
[RxBLEKit|VERB|13:46:34.660]: CentralManager(10775823040) didDiscover(peripheral: Peripheral(uuid: FDB44F24-E04B-541D-1937-B56047DD700B, name: nil),
[RxBLEKit|VERB|13:46:34.667]: CentralManager(10775823040) didDiscover(peripheral: Peripheral(uuid: B83CAAC7-A78B-CF18-B4B5-8B2D71BA337C, name: nil),
[RxBLEKit|DEBG|13:46:34.670]: CentralManager(10775823040) stopScan()
[RxBLEKit|DEBG|13:46:34.673]: CentralManager(10775823040) retrievePeripherals(
[RxBLEKit|DEBG|13:46:34.677]: CentralManager(10775823040) connect(
[RxBLEKit|DEBG|13:46:34.940]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverServices(
[RxBLEKit|DEBG|13:46:35.395]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverServices(services
[RxBLEKit|DEBG|13:46:35.406]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverCharacteristics(
[RxBLEKit|DEBG|13:46:35.407]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverCharacteristics(
[RxBLEKit|DEBG|13:46:35.408]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverCharacteristics(
[RxBLEKit|DEBG|13:46:35.516]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverCharacteristicsFor(for:
[RxBLEKit|DEBG|13:46:35.634]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverCharacteristicsFor(for:
[RxBLEKit|DEBG|13:46:35.757]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverCharacteristicsFor(for:
[RxBLEKit|DEBG|13:46:35.854]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) writeValue(
[RxBLEKit|DEBG|13:46:35.856]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverServices(
[RxBLEKit|DEBG|13:46:35.903]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didWriteValueFor(for:Characteristic(uuid: 5B7DD565-6E90-4DF8-B806-F4AC85DF74CA, id: 10766643872),
[RxBLEKit|DEBG|13:46:35.904]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverServices(services
[RxBLEKit|DEBG|13:46:35.905]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverCharacteristics(
[RxBLEKit|DEBG|13:46:35.906]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverCharacteristics(
[RxBLEKit|DEBG|13:46:35.906]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverCharacteristics(
[RxBLEKit|DEBG|13:46:35.907]: Periph[RxBLEKit|DEBG|13:46:35.907]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscove[RxBLEKit|DEBG|13:46:35.908]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverCharacteristicsFor(for:
[RxBLEKit|DEBG|13:46:39.436]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) setNotifyValue(
enabled: true,
[RxBLEKit|DEBG|13:46:39.439]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didUpdateNotificationStateFor(
[RxBLEKit|DEBG|13:46:39.442]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) setNotifyValue(
enabled: false,
[RxBLEKit|DEBG|13:46:39.444]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didUpdateNotificationStateFor(
[RxBLEKit|DEBG|13:46:40.453]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) setNotifyValue(
enabled: true,
[RxBLEKit|DEBG|13:46:40.455]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didUpdateNotificationStateFor(
[RxBLEKit|DEBG|13:46:40.458]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) setNotifyValue(
enabled: false,
[RxBLEKit|DEBG|13:46:40.460]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didUpdateNotificationStateFor(
`

Seems that it is always setting notification to false after setting it to true. Any ideas?

packetstracer, 2019-05-22:
This is a simplified (code just trying to monitor once) one:

`
[RxBLEKit|DEBG|14:00:49.308]: CentralManager(10861909952) didUpdateState(state: poweredOn)
[RxBLEKit|DEBG|14:00:49.357]: CentralManager(10861909952) scanForPeripherals(
[RxBLEKit|VERB|14:00:49.390]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: 45D92E51-0493-CBB1-1EAD-C0FEC42A38F0, name: nil),
[RxBLEKit|VERB|14:00:49.394]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: 620C7A60-5B38-365F-882C-207C6CF9D44D, name: nil),
[RxBLEKit|VERB|14:00:49.395]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: D1B8824B-E213-EE20-5A8A-2E89FB12498E, name: nil),
[RxBLEKit|VERB|14:00:49.396]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: 2A06BC7B-788B-3A28-8D9B-2806F220408A, name: nil),
[RxBLEKit|VERB|14:00:49.399]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: 57605792-F730-B13E-78F1-F152DEA72984, name: nil),
[RxBLEKit|VERB|14:00:49.400]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: 203141B1-131A-2B24-6E67-CDF3C562C65D, name: nil),
[RxBLEKit|VERB|14:00:49.405]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")),
[RxBLEKit|VERB|14:00:49.406]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")),
[RxBLEKit|VERB|14:00:49.420]: CentralManager(10861909952) didDiscover(peripheral: Peripheral(uuid: 3AEFCE6D-2EE0-05F5-6B31-AFB2F1CE64DB, name: nil),
[RxBLEKit|DEBG|14:00:49.424]: CentralManager(10861909952) stopScan()
[RxBLEKit|DEBG|14:00:49.429]: CentralManager(10861909952) retrievePeripherals(
[RxBLEKit|DEBG|14:00:49.434]: CentralManager(10861909952) connect(
[RxBLEKit|DEBG|14:00:49.783]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverServices(
[RxBLEKit|DEBG|14:00:50.225]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverServices(services
[RxBLEKit|DEBG|14:00:50.238]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverCharacteristics(
[RxBLEKit|DEBG|14:00:50.239]: Periph
[RxBLEKit|DEBG|14:00:50.240]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) discoverCharacteristics(
[RxBLEKit|DEBG|14:00:50.344]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverCharacteristicsFor(for:
[RxBLEKit|DEBG|14:00:50.463]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverCharacteristicsFor(for:
[RxBLEKit|DEBG|14:00:50.584]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didDiscoverCharacteristicsFor(for:
[RxBLEKit|DEBG|14:00:50.634]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) setNotifyValue(enabled: true,
[RxBLEKit|DEBG|14:00:50.636]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didUpdateNotificationStateFor(
[RxBLEKit|DEBG|14:00:50.637]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) setNotifyValue(enabled: false,
[RxBLEKit|DEBG|14:00:50.639]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didUpdateNotificationStateFor(
`

packetstracer, 2019-05-24:
Tried to read and write on that and other characteristics and works fine, so it seems it's not a problem of service/characteristic definition.

dariuszseweryn, 2019-05-24:
Why do you think so?

packetstracer, 2019-05-24:
Because characteristics is published, seems to have the correct values for notification, and other programs can subscribe to the characteristic notification normally

dariuszseweryn, 2019-05-24:
Yet you can see here that iOS API rejects setting the notification:
```
[RxBLEKit|DEBG|14:09:38.284]: Peripheral(uuid: 80293EA3-F380-A86F-B84F-DFC1A5B23502, name: Optional("XFit")) didUpdateNotificationStateFor(for:Characteristic(uuid: D8B2C95B-B317-47CA-AC02-CCFD3D85A666, id: 11081319456), isNotifying: false, error: Optional(Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo={NSLocalizedDescription=The attribute could not be found.}))
```

What other programs have you tried?

packetstracer, 2019-05-24:
LightBlue

packetstracer, 2019-05-24:
If I disable in the iOS project the line that throws the error it works properly.

Peripheral.swift (line 334)

```swift
    public func setNotifyValue(_ enabled: Bool,
                               for characteristic: Characteristic) -> Observable<Characteristic> {
        let observable = peripheral
            .rx_didUpdateNotificationStateForCharacteristic
            .filter { $0.0 == characteristic.characteristic }
            .take(1)
            .map { (_, error) -> Characteristic in
                if let error = error {
                    /// throw BluetoothError.characteristicNotifyChangeFailed(characteristic, error)
                }
                return characteristic
            }
        return ensureValidPeripheralStateAndCallIfSucceeded(
            for: observable,
            postSubscriptionCall: { [weak self] in
                self?.peripheral.setNotifyValue(enabled, for: characteristic.characteristic)
            }
        )
    }
```

packetstracer, 2019-05-24:
I'll investigate the characteristic definition on the peripheral side. The current characteristic definition on the Android device is: 

```c
  pCharSnifferVoltage = pService->createCharacteristic(
    SNIFFER_VOLTAGE_UUID,
    BLECharacteristic::PROPERTY_READ |
    BLECharacteristic::PROPERTY_NOTIFY |
    BLECharacteristic::PROPERTY_WRITE
  );
```

dariuszseweryn, 2019-05-24:
So _maybe_ there is some error in the implementation of ESP32 you are using after all?

packetstracer, 2019-05-24:
Could be, I'll post on Espressif forums in case someone has encountered the same problem. I'll post my findings into this thread.

In order to not keep maintaining my own react-native-ble-plx fork, is there a way (config or something) to make the library not stop native(iOS) exceptions?

Cierpliwy, 2019-05-24:
Just to be sure. Is Client Characteristic Configuration Descriptor present for your characteristic? You can check it with nRF Connect app for example.

asurare, 2019-05-24:
@Cierpliwy I had a similar issue (https://github.com/Polidea/react-native-ble-plx/issues/229) and I simply did what you wrote above 
`/// throw BluetoothError.characteristicNotifyChangeFailed(characteristic, error)`

I am using a custom bluetooth peripheral, so it mights follow the requirements of the bluetooth protocol. If you wonder what's this for @packetstracer, you should look at this post https://devzone.nordicsemi.com/f/nordic-q-a/8427/why-do-we-need-cccd

packetstracer, 2019-05-27:
> Just to be sure. Is Client Characteristic Configuration Descriptor present for your characteristic? You can check it with nRF Connect app for example.

@Cierpliwy I've investigated about the Client Characteristic Configuration descriptor for the characteristic that was notifying, and It wasn't set. So I've created the descriptor and added to the characteristic and now everything works fine. 

Thanks a lot!! 

For the sake of future information for people having the same issue, I attach some info about Client Characteristic Configuration descriptor:

What is this descriptor:

> Associated with the idea of indications/notifications, is the architected BLE Descriptor called "Client Characteristic Configuration" which has UUID 0x2902. This contains (among other things) two distinct bit fields that can be on or off. One bit field governs Notifications while the other governs Indications. If the corresponding bit is on, then the server can/may send the corresponding push. For example if the Notifications bit is on, then the server can/may send notifications. The primary purpose of the descriptor is to allow a partner (a client) to request that the server actually send notifications or indications. Here is an example.
> Imagine that we have a BLE Server that can publish notifications when data changes. Now imagine that a BLE Client connects but is actually not interested in receiving these. If the BLE Server executes a notification push and sends the data, that will be wasted effort/energy as the client doesn't want or can't use the information. What we really want is that the client should inform the server that if it wants to push new data then it either should or shouldn't. And that is where this descriptor comes into play.
> If the descriptor is present on a characteristic then then client can remotely change the bit flag to enable or disable notifications and indications. Since the descriptor flag is stored locally within the server, the server is at liberty to examine the flag before performing a radio transmission. What this means is that the client/peer can toggle on or off its desire to receive notifications and the server should honor those requests.

Extracted from [https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf](https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf)

How I created this descriptor in Arduino for the characteristic that notifies de values:

```c
  BLEService *pService = pServer->createService(SERVICE_SNIFFER_UUID);

  BLECharacteristic *pCharSnifferVoltage = pService->createCharacteristic(
    SNIFFER_VOLTAGE_UUID,
    BLECharacteristic::PROPERTY_NOTIFY
  );

  // create CCC descriptor and add it to the characteristic
  BLEDescriptor *pClientCharacteristicConfigDescriptor = new BLEDescriptor((uint16_t)0x2902);
  pCharSnifferVoltage->addDescriptor(pClientCharacteristicConfigDescriptor);
```

closed 2019-05-27 by packetstracer

sushrut-desora, 2020-08-12:
> If I disable in the iOS project the line that throws the error it works properly (notification handler executes whenever the peripheral broadcasts a value, and the characteristic has that value).
> 
> Peripheral.swift (line 334)
> 
> ```swift
>     public func setNotifyValue(_ enabled: Bool,
>                                for characteristic: Characteristic) -> Observable<Characteristic> {
>         let observable = peripheral
>             .rx_didUpdateNotificationStateForCharacteristic
>             .filter { $0.0 == characteristic.characteristic }
>             .take(1)
>             .map { (_, error) -> Characteristic in
>                 if let error = error {
>                     /// throw BluetoothError.characteristicNotifyChangeFailed(characteristic, error)
>                 }
>                 return characteristic
>             }
>         return ensureValidPeripheralStateAndCallIfSucceeded(
>             for: observable,
>             postSubscriptionCall: { [weak self] in
>                 self?.peripheral.setNotifyValue(enabled, for: characteristic.characteristic)
>             }
>         )
>     }
> ```

This works well to resolve the issue. For 1.x version of react-native-ble-plx i created a [patch-package](https://www.npmjs.com/package/patch-package) patch file and would apply the patch 'postinstall'. 
However, in react-native-ble-plx 2.x they seem to use 'MultiplatformBleAdapter' as a pod dependency and i am unable to create a patch. I do not have access to the ble firmware so I can only deal with the issue on app. Is there a library which could help me generate and apply patch file to a swift file within pods folder rather than editing the file manually ?

richmouritsen, 2020-12-04:
I was running into this same problem and added the CCCD to the notify characteristic as recommended above. Now, instead of getting the `the attribute is not found` native error, I am getting a `the handle is invalid` error. I would prefer to not change the native code as a workaround. Any ideas what would be wrong with my descriptor that would cause be to get a "the handle is invalid" error?

dariuszseweryn, 2020-12-04:
You hav probably misconfigured your peripheral. Consult Bluetooth Core Specification and/or documentation of the framework you use for peripheral implementation.

nolife08021, 2021-02-22:
@dariuszseweryn  Hi, i am facing the same issue. It working on android, but failed on IOS.

{"message":"Characteristic 000036f6-0000-1000-8000-00805f9b34fb notify change failed for device A and service 0000fee7-0000-1000-8000-00805f9b34fb","errorCode":403,"attErrorCode":10,"iosErrorCode":null,"androidErrorCode":null,"reason":"The attribute could not be found.","name":"BleError"}

nolife08021, 2021-02-22:
> If I disable in the iOS project the line that throws the error it works properly (notification handler executes whenever the peripheral broadcasts a value, and the characteristic has that value). [...]

Any workaround for v2 ?

ameriles, 2021-03-06:
> Any workaround for v2 ?

You can:
1. Open workspace with XCode 
2. Expand Pods project
3. Open Pods folder
4. Look for MultiplatformBleAdapter folder and open it
5. Look for the Peripheral.swift file, make the change and build

Awful solution I know, but works for a workaround until we can have a better way

t3db0t, 2021-09-01:
I just want to chime in here and say that @ameriles 's above solution also worked for me. I have no reason to believe that the Bluetooth device I'm working with (Digi XBee 3 Pro) doesn't rigorously follow the Bluetooth spec. Perhaps the "Characteristic notify changed failed" error should be a warning instead?

ChaossAdept7, 2021-11-30:
I have the same issue....

fosgate1, 2021-12-11:
Confirmed that @ameriles solution worked.   You can automate the patch when the script below. 

`
#!/usr/bin/env bash

if grep -qF "/// throw BluetoothError.characteristicNotifyChangeFailed(characteristic, error)" ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/Peripheral.swift; then
    echo "MultiplatformBleAdapter already patched"
else
    echo "Patching MultiplatformBleAdapter"
    OLD='throw BluetoothError.characteristicNotifyChangeFailed(characteristic, error)'
    NEW='\/\/\/ throw BluetoothError.characteristicNotifyChangeFailed(characteristic, error)'
    sed -ie "s/${OLD}/${NEW}/" ios/Pods/MultiplatformBleAdapter/iOS/RxBluetoothKit/Peripheral.swift
    echo "Done"
fi
`

keerthi552, 2021-12-31:
const SERVICE_SNIFFER_UUID = 'ef66ccad-f4a8-4c51-b025-bc2a2c9fdaa6'
  const SNIFFER_STATUS_UUID = '5b7dd565-6e90-4df8-b806-f4ac85df74ca'
  const SNIFFER_SPEED_UUID = 'c8fb3a51-d44c-4d9f-a8ec-a7598c1bf2ea'
  const SNIFFER_VOLTAGE_UUID = 'd8b2c95b-b317-47ca-ac02-ccfd3d85a666'

may i know how can we get these uuid's

chriswgreen-lineup, 2022-03-18:
Just want to highlight the fix by [packetstracer](https://github.com/packetstracer). 

ADD A DESCRIPTOR:

```
BLEDescriptor *pClientCharacteristicConfigDescriptor = new BLEDescriptor((uint16_t)0x2902);
pOutputChar->addDescriptor(pClientCharacteristicConfigDescriptor);
```

KthProg, 2024-02-07:
> Just want to highlight the fix by [packetstracer](https://github.com/packetstracer).
> 
> ADD A DESCRIPTOR:
> 
> ```
> BLEDescriptor *pClientCharacteristicConfigDescriptor = new BLEDescriptor((uint16_t)0x2902);
> pOutputChar->addDescriptor(pClientCharacteristicConfigDescriptor);
> ```

Didn't fix the issue for me.

Rubon72, 2025-04-10:
> > Just to be sure. Is Client Characteristic Configuration Descriptor present for your characteristic? You can check it with nRF Connect app for example.
> 
> [@Cierpliwy](https://github.com/Cierpliwy) I've investigated about the Client Characteristic Configuration descriptor for the characteristic that was notifying, and It wasn't set. So I've created the descriptor and added to the characteristic and now everything works fine.
> 
> Thanks a lot!!
> 
> For the sake of future information for people having the same issue, I attach some info about Client Characteristic Configuration descriptor:
> 
> What is this descriptor:
> 
> > Associated with the idea of indications/notifications, is the architected BLE Descriptor called "Client Characteristic Configuration" which has UUID 0x2902. This contains (among other things) two distinct bit fields that can be on or off. One bit field governs Notifications while the other governs Indications. If the corresponding bit is on, then the server can/may send the corresponding push. For example if the Notifications bit is on, then the server can/may send notifications. The primary purpose of the descriptor is to allow a partner (a client) to request that the server actually send notifications or indications. Here is an example.
> > Imagine that we have a BLE Server that can publish notifications when data changes. Now imagine that a BLE Client connects but is actually not interested in receiving these. If the BLE Server executes a notification push and sends the data, that will be wasted effort/energy as the client doesn't want or can't use the information. What we really want is that the client should inform the server that if it wants to push new data then it either should or shouldn't. And that is where this descriptor comes into play.
> > If the descriptor is present on a characteristic then then client can remotely change the bit flag to enable or disable notifications and indications. Since the descriptor flag is stored locally within the server, the server is at liberty to examine the flag before performing a radio transmission. What this means is that the client/peer can toggle on or off its desire to receive notifications and the server should honor those requests.
> 
> Extracted from https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf
> 
> How I created this descriptor in Arduino for the characteristic that notifies de values:
> 
>   BLEService *pService = pServer->createService(SERVICE_SNIFFER_UUID);
> 
>   BLECharacteristic *pCharSnifferVoltage = pService->createCharacteristic(
>     SNIFFER_VOLTAGE_UUID,
>     BLECharacteristic::PROPERTY_NOTIFY
>   );
> 
>   // create CCC descriptor and add it to the characteristic
>   BLEDescriptor *pClientCharacteristicConfigDescriptor = new BLEDescriptor((uint16_t)0x2902);
>   pCharSnifferVoltage->addDescriptor(pClientCharacteristicConfigDescriptor);
> 
>   pService->start();

thank you. Дзякуй братка