dotintent/react-native-ble-plx #469: Bluetooth pairing with a password / pin

jinqiangg, 2019-05-22:
I am trying to connect a bluetooth device but it requires a pin. Is there any such feature to connect the device with pin?

Cierpliwy, 2019-05-24:
Pairing procedure is mostly handled by OS. Do you use Android or iOS?

NastyZ98, 2019-06-19:
Hello @jinqiangg if you have successfully paired your device with a password/pin can you describe the procedure please ?

Thank you,
Nathan.

renard12, 2019-08-23:
@Cierpliwy 
Hi, so I am using Android 7 and at this OS we have no pairing automatically. 
My code works without issues on Android 8 and higher, so I need to bound manually

tangtai, 2019-09-12:
I am having similar problem, the device connect requires PIN to establish connection,  the device can be authenticated in bluetooth settings in the android os, but i want the establish connection in ble-plx. i konw you can trigger authentication in application,  i wonder if ble-plx can do it. and if i establish connection using the system, can i still receive data in my bleManager,

closed 2019-11-18 by stale[bot]

WarishaAli, 2020-04-05:
Hi. I'd also like to connect my android with a bluetooth device without any user input, i.e. no pairing. possible using this library, or possible at all in the android OS?

aarthigovardhanan88, 2021-05-27:
Any inputs on this?

oskarhertzman, 2022-05-19:
maybe the creators could be useful for once and answer this thread?