dotintent/react-native-ble-plx #471: Device.monitorCharacteristicForService() listener always return an error when device is connected

Techno-Hwizrdry, 2019-05-24:
Greetings!  I'm learning how to use react-native-ble-plx with redux-sagas by playing around with the [sagas in the example App](https://github.com/Cierpliwy/SensorTag/blob/master/src/Saga.js).  I'm trying to add code that will enable notifications [from inside function* handleConnection] and update a state in my redux store.  After reading [this article on redux-saga event channels](https://medium.com/@emaldonado.daniel/redux-saga-event-channel-8bc4bdf4567c) and looking at the other parts of src/Saga.js, it made sense to me to use eventChannel like so.

```javascript
function* handleConnection(manager: BleManager): Generator<*, *, *> {
  var testTask = null

  for (;;) {
    // Take action
    const { device }: ConnectAction = yield take("CONNECT")
    const disconnectedChannel = /*...disconnectedChannel code here...*/
    const setupNotificationChannel = yield eventChannel((emit) => {
      const subscription = device.monitorCharacteristicForService(SERVICE_UUID, CHAR_UUID, (error, characteristic) => {
          emit([error, characteristic])
      })
      return () => {
        subscription.remove()
      }
    }, buffers.expanding(1))
```
Then, within the if(deviceAction) block of the for(;;) loop, I handle the characteristic or error:

```javascript
for (;;) {
        const { deviceAction, disconnected } = yield race({
          deviceAction: take(deviceActionChannel),
          disconnected: take(disconnectedChannel)
        })

        if (deviceAction) {
          if (deviceAction.type == "SETUP_NOTIFY_BUTTONVALUE") {
            yield put(log("Setting up notifications..."))

            const [error, characteristic]: [?BleError, ?Characteristic] = yield take(setupNotificationChannel)

            if (error != null) {
              console.warn("Device name = " + device.localName + "  Device ID = " + device.id)
              console.warn(error.message)
            }

            if (characteristic != null) {
              var new_Button_state = yield select(selectors.getButtonState)
              new_Button_state.button_value = characteristic.value
              yield put(updateButtonState(new_Button_state))
            }
          }
```

However, my listener function for device.monitorCharacteristicForService() is always emitting an error and a null characteristic.  error.message always come back with "Device id is not connected", even if a successful connection has already been established with the device.  This occurs when I run it on both iOS and Android.

Is this a result of my misuse of redux-saga or is this a greater issue device.monitorCharacteristicForService()?  Any help will be greatly appreciated.

Cheers!

dariuszseweryn, 2019-05-26:
> Is this a result of my misuse of redux-saga or is this a greater issue device.monitorCharacteristicForService()?

I would expect the former. I see no code related to connecting to the device. And so does the error message state. So you probably did not connect.

Techno-Hwizrdry, 2019-05-27:
I forgot to mention (my fault) in my original post that I left the code, in Saga.js, for connecting to the device unchanged.  Like so:

```javascript
try {
      yield put(updateConnectionState(ConnectionState.CONNECTING))
      yield call([device, device.connect])
      yield put(updateConnectionState(ConnectionState.DISCOVERING))
      yield call([device, device.discoverAllServicesAndCharacteristics])
      yield put(updateConnectionState(ConnectionState.CONNECTED))

      for (;;) {
        const { deviceAction, disconnected } = yield race({
          deviceAction: take(deviceActionChannel),
          disconnected: take(disconnectedChannel)
        })

        if (deviceAction) {
          if (deviceAction.type == "SETUP_NOTIFY_BUTTONVALUE") {
            yield put(log("Setting up notifications..."))

            const [error, characteristic]: [?BleError, ?Characteristic] = yield take(setupNotificationChannel)

            if (error != null) {
              console.warn("Device name = " + device.localName + "  Device ID = " + device.id)
              console.warn(error.message)
            }

            if (characteristic != null) {
              var new_Button_state = yield select(selectors.getButtonState)
              new_Button_state.button_value = characteristic.value
              yield put(updateButtonState(new_Button_state))
            }
          }
```

The strange thing is, I am able to connect and read from that characteristic and on the device (via device.readCharacteristicForService). So I must be successfully connecting to it.

dariuszseweryn, 2019-05-28:
https://github.com/Polidea/react-native-ble-plx/issues/468

closed 2019-08-03 by stale[bot]