dotintent/react-native-ble-plx #472: Question: I need connected device name, What changes need?

Karanp13, 2019-05-25:
Initialisation
```
constructor(props) {
    super(props);
    this.manager = new BleManager();
}
```
Start Subscription
```
componentWillMount() {
    const subscription = this.manager.onStateChange((state) => {
        if (state === 'PoweredOn') {
            this.scanAndConnect();
            subscription.remove();
        }
    }, true);
  }
```
Find Device Name
```
  scanAndConnect() {
    this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            // Handle error (scanning will be stopped automatically)
            return
        }
        var value = device.name;
        this.setState({bluetoothDevicename : value})
        // Check if it is a device you are looking for based on advertisement data
        // or other criteria.
        //if (device.name === 'TI BLE Sensor Tag' || 
          //  device.name === 'SensorTag') {   
            // Stop scanning as it's not necessary if you are scanning for one device.
            this.manager.stopDeviceScan();
            // Proceed with connection.
        //}
    });
}
```

keyz23, 2019-05-28:
You need to connect to a specif device name?? 
You need to  uncommented this part:
```

//if (device.name === 'TI BLE Sensor Tag' || 
          //  device.name === 'SensorTag')
```
and insert your device name.

Karanp13, 2019-05-28:
> You need to connect to a specif device name??
> You need to uncommented this part:
> 
> ```
> 
> //if (device.name === 'TI BLE Sensor Tag' || 
>           //  device.name === 'SensorTag')
> ```
> 
> and insert your device name.

I don't need specific, Any device connect to phone then show the device name, Specific works well I tried.

closed 2019-08-03 by stale[bot]