dotintent/react-native-ble-plx #477: Question

psambit9791, 2019-05-29:
- [Yes] I am running the latest version
- [Yes] I checked the documentation and found no answer
- [Yes] I checked to make sure that this issue has not already been filed
- [Yes] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general.

My objective is to write notification enable/disable values to 0x002902, but it has multiple handles within it for various functions. This is what the UUID looks like on gatttool:

char-read-uuid 00002902-0000-1000-8000-00805f9b34fb
handle: 0x0004 	 value: 00 00 
handle: 0x0023 	 value: 00 00 
handle: 0x0026 	 value: 00 00 
handle: 0x0029 	 value: 00 00 
handle: 0x0034 	 value: 00 00 

I cannot find any information on the documentation on how to write values to these specific handles within the UUID.

Any help would be much appreciated.

dariuszseweryn, 2019-05-29:
This does not look like a question regarding this library but a general bluetooth question. This library does not expose any handles. Are you sure you wanted to ask the question here?

psambit9791, 2019-05-29:
I tried using the following code to write value to the specific UUID:

const characteristic = await this.device.writeCharacteristicWithResponseForService(SERVICE_UUID, "00002902-0000-1000-8000-00805f9b34fb", value);

But since the UUID contains multiple attributes, the write could not take place. Is there a way to specify how I can write to the specific attribute of a characteristic?

dariuszseweryn, 2019-05-29:
There is no support for writing descriptors in this library yet. It is all handled for you. https://github.com/Polidea/react-native-ble-plx/wiki/Characteristic-Notifying

psambit9791, 2019-05-29:
Thanks. That worked!!!

closed 2019-05-29 by psambit9791

mvmmarcus, 2021-08-04:
@psambit9791 , im trying to do this. How you do to get notifications for this descriptor ?

psambit9791, 2021-08-04:
This was a long time back. From memory, the monitor() function on the relevant characteristicUUID helped me.

mvmmarcus, 2021-08-04:
> This was a long time back. From memory, the monitor() function on the relevant characteristicUUID helped me.

by monitoring the feature I can get all the data when I write to the RACP feature, but I can't get the data when I send it directly from the device to my app. Searching a lot I found it might be the descriptor 2902 that I wasn't dealing with. But I don't know how to monitor this descriptor :(

mvmmarcus, 2021-08-04:
currently this is what I have. I can read all meter data, but I cannot send meter data to my app. This way I can't get the glucose value when this data is measured

`import React from 'react';
import {View, Text, FlatList, Button, StyleSheet} from 'react-native';
import useStream from './useStream';
import adapterStateStream from './adapterStateStream';
import deviceStream from './deviceStream';
import {Buffer as buffer} from 'buffer';
import {BleManager} from 'react-native-ble-plx';
const bleManager = new BleManager();

const connect = async deviceId => {
  const device = await bleManager.connectToDevice(deviceId);
  console.log('connected device: ', device);
  await device.discoverAllServicesAndCharacteristics();

  const services = await device.services();
  console.log('services: ', services);
};

const notifyCharacteristic = (deviceId, characteristicUUID) => {
  const serviceUUID = '00001808-0000-1000-8000-00805f9b34fb';

  bleManager.monitorCharacteristicForDevice(
    deviceId,
    serviceUUID,
    characteristicUUID,
    async (error, characteristic) => {
      if (error) {
        console.log('monitorError: ', error);
      }

      // console.log('monitorCharacteristic: ', characteristic);
      // // console.log('monitorCharacteristicValue: ', characteristic.value);
      console.log('monitorCharacteristic: ', characteristic);
      const values = buffer.from(characteristic?.value || '', 'base64');
      console.log('values: ', values);
      if (values.length > 5) {
        const glucoseValue = values[12]; // 12th position in the case of my device
        console.log('glucoseValue: ', glucoseValue);
      }
    },
  );
};

const writeCharacteristic = async (
  deviceId,
  characteristicUUID,
  command = '0x0101',
) => {
  const serviceUUID = '00001808-0000-1000-8000-00805f9b34fb';

  // const dataMostSignificantByte = (command >> 8) & 0xff;
  // const dataLeastSignificantByte = command & 0xff;

  // const dataByteArrayInLittleEndian = [
  //   dataLeastSignificantByte,
  //   dataMostSignificantByte,
  // ];
  // const dataInBase64 = buffer
  //   .from(dataByteArrayInLittleEndian)
  //   .toString('base64');

  const allValuesRetrieverInBase64 = buffer
    .from([0x01, 0x06])
    .toString('base64');

  await bleManager.writeCharacteristicWithResponseForDevice(
    deviceId,
    serviceUUID,
    characteristicUUID,
    allValuesRetrieverInBase64,
  );
};

const App = () => {
  const adapterState = useStream(adapterStateStream, '');
  const devices = useStream(deviceStream, {});

  return (
    <View>
      <Text>Adapter State: {adapterState}</Text>
      <FlatList
        style={styles.flatLsit}
        data={Object.values(devices).slice(0, 3)}
        renderItem={({item}) => (
          <>
            <Button
              title={item.name || item.id}
              onPress={() => connect(item.id)}
            />
            <Button
              title={'NotiFy Glucose Measurements'}
              onPress={() =>
                notifyCharacteristic(
                  item.id,
                  '00002a18-0000-1000-8000-00805f9b34fb',
                )
              }
            />
            <Button
              title={'NotiFy Glucose Measurements Context'}
              onPress={() =>
                notifyCharacteristic(
                  item.id,
                  '00002a34-0000-1000-8000-00805f9b34fb',
                )
              }
            />
            <Button
              title={'NotiFy RACP'}
              onPress={() =>
                notifyCharacteristic(
                  item.id,
                  '00002a52-0000-1000-8000-00805f9b34fb',
                )
              }
            />

            <Button
              onPress={() =>
                writeCharacteristic(
                  item.id,
                  '00002a52-0000-1000-8000-00805f9b34fb',
                  '0x0101',
                )
              }
              title={'Write To RACP'}
            />
          </>
        )}
        keyExtractor={device => device.id}
      />
    </View>
  );
};

const styles = StyleSheet.create({
  collision: {
    fontSize: 100,
    textAlign: 'center',
  },
  button: {
    fontSize: 70,
    width: '100%',
    textAlign: 'center',
  },
  buttonHalf: {
    fontSize: 70,
    width: '50%',
    textAlign: 'center',
  },
  row: {
    flexDirection: 'row',
  },
});

export default App;
`