dotintent/react-native-ble-plx #478: Device disconnects on monitorCharacteristicForDevice

psambit9791, 2019-06-01:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ Yes ] I am running the latest version
- [ Yes ] I checked the documentation and found no answer
- [ Yes ] I checked to make sure that this issue has not already been filed
- [ Yes ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general.

I have been trying to monitor characteristics for a device on a specific characteristic. The objective is to initiate a subscription to the Delete Characteristic. Then start the delete process by writing to the delete characteristic. Once the deletion is done, the subscription notifies me with a value.

This is my code:

```
CHARACTERISTIC_UUID = {
DELETE:  "04b30006-6f01-4987-bf0e-f8aecc7b42a8",
}

async setupNotificationForDelete() {
        deviceID = this.device.id;
        console.log(deviceID);
        this.delete_notif = await this.manager.monitorCharacteristicForDevice(deviceID, SERVICE_UUID, CHARACTERISTIC_UUID.DELETE.write, (error, characteristic) => {
            if (error) {
                console.log(error);
                return
            };
            delete_complete = convertBase64toHex(characteristic.value);
            if (delete_complete == '00') {
                console.log(delete_complete, "DELETE COMPLETE");
            }
        });
    }

deleteData = () => {
        console.log("Delete Started");
        this.device.writeCharacteristicWithResponseForService(SERVICE_UUID, CHARACTERISTIC_UUID.DELETE.write, convertHexToBase64('01')).then(() => {
            console.log('Completed Writing');
        });
    }

async handlePressDeleteData() {
        console.log("Delete Data");
        this.device.isConnected().then((value) => {this.connectionState = value});
        await this.setupNotificationForDelete();
        this.deleteData();
    }
```

Whenever I try call handlePressDelete(), the device automatically disconnects and shows as a BLEerror.

Can anyone tell me why this has been happening?

Also, use nRF Connect to see CCCD linked to the characteristics, it does exist:

[image]

dariuszseweryn, 2019-06-01:
What does the error tell you? Have you examined it?

psambit9791, 2019-06-01:
The error as shown on the console just says:

`06-01 01:27:19.646 26685 26736 I ReactNativeJS: [BleError: Device 08:6B:D7:41:BA:79 was disconnected]`

followd by:

```
06-01 01:27:21.624 26685 26736 W ReactNativeJS: Possible Unhandled Promise Rejection (id: 0):
06-01 01:27:21.624 26685 26736 W ReactNativeJS: BleError: Characteristic 04b30006-6f01-4987-bf0e-f8aecc7b42a8 write failed for device 08:6B:D7:41:BA:79 and service 04b30000-6f01-4987-bf0e-f8aecc7b42a8
```

This causes other promises to fail (which is when I identified the problem) but on printing the error, this is all that shows up.

dariuszseweryn, 2019-06-01:
Have you examined something more than only the message? I.e. did you print all the properties the error has?

psambit9791, 2019-06-01:
I have done a console.log(error) for the error as returned from the monitorCharacteristicForDevice(). I believe that does print all the properties.
That prints only this:

`06-01 01:27:19.646 26685 26736 I ReactNativeJS: [BleError: Device 08:6B:D7:41:BA:79 was disconnected]`

dariuszseweryn, 2019-06-01:
I believe you are wrong. Try `console.log(JSON.stringify(error))` and check it out.

psambit9791, 2019-06-01:
Yeah. My bad. 
The complete error message looks like this:

`06-01 18:47:37.941 29744 29797 I ReactNativeJS: {"message":"Device 08:6B:D7:41:BA:79 was disconnected","errorCode":201,"attErrorCode":8,"iosErrorCode":null,"androidErrorCode":null,"reason":"Disconnected from 08:6B:D7:41:BA:79 with status 8 (GATT_INSUF_AUTHORIZATION)","name":"BleError","line":86743,"column":107,"sourceURL":"http://localhost:8081/index.delta?platform=android&dev=true&minify=false"}`

As I understand a GATT_INSUF_AUTHORIZATION implies that the value cannot be written onto the device. But, in my case, the value as specified in:
```
this.device.writeCharacteristicWithResponseForService(SERVICE_UUID, CHARACTERISTIC_UUID.DELETE, convertHexToBase64('01')).then(() => {
            console.log('Completed Writing');
        });
```
does get written on to the device, but the disconnect happens after that.

Any ideas on that?

dariuszseweryn, 2019-06-01:
Yes. The device tells that there is insufficient authorization or there may be a simple timeout (both issues share the same error code in Android API) for whatever reason e.g. peripheral reset

psambit9791, 2019-06-01:
I updated the code a little bit and it now looks like this:

```

async setupNotificationForDelete() {
        deviceID = this.device.id;
        console.log("Notification activated", this.device.id);
        await this.device.writeCharacteristicWithResponseForService(SERVICE_UUID, CHARACTERISTIC_UUID.DELETE.write, convertHexToBase64('01')).then(() => {
            console.log('Completed Writing');
        });
        console.log("Notification2", this.device.id);
        this.delete_notif = this.manager.monitorCharacteristicForDevice(deviceID, SERVICE_UUID, CHARACTERISTIC_UUID.DELETE.write, (error, characteristic) => {
            if (error) {
                console.log(JSON.stringify(error));
                return
            };
            console.log("DelNotifReached", characteristic.value);
            // delete_complete = convertBase64toHex(characteristic.value);
            // if (delete_complete == '00') {
            //     console.log(delete_complete, "DELETE COMPLETE");
            //     this.handleButtonStateDefault();
            //     //this.delete_notif.remove()
            // }
        });
    }

handlePressDeleteData = () => {
    this.setupNotificationForDelete();
}
```

But now , the error shows up as this:

`06-01 21:12:07.653 15001 15052 I ReactNativeJS: {"message":"Characteristic 04b30006-6f01-4987-bf0e-f8aecc7b42a8 notify change failed for device ? and service 04b30000-6f01-4987-bf0e-f8aecc7b42a8","errorCode":403,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":"Cannot write client characteristic config descriptor (code 3) with characteristic UUID 04b30006-6f01-4987-bf0e-f8aecc7b42a8","name":"BleError","line":86743,"column":107,"sourceURL":"http://localhost:8081/index.delta?platform=android&dev=true&minify=false"}`

Would be great if you could help. Cannot find any headway as to why this is happening.

dariuszseweryn, 2019-06-03:
I would need logs from the native side to say more. You had a point about it the template and/or in documentation

psambit9791, 2019-06-03:
Logged the Native Side. Adding them here.

The unhandled promise rejection shows up at the very end but I cannot seem to find any Exceptions occurring in this context.

```
06-03 10:55:49.750  3255  3790 D InputReader: Input event(11): value=1 when=353453.443975
06-03 10:55:49.750  3255  3790 I InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.2845 ] when=353453.443975
06-03 10:55:49.750  3255  3789 I InputDispatcher: Delivering touch to (5250): action: 0x4, displayId: 0, '8445d9 ', toolType: 1
06-03 10:55:49.750  3255  3789 I InputDispatcher: Delivering touch to (5250): action: 0x4, displayId: 0, '42ee231', toolType: 1
06-03 10:55:49.750  3255  3789 I clp-JNI : Add pinfo) 0
06-03 10:55:49.751  3255  3789 I InputDispatcher: Delivering touch to (20634): action: 0x0, displayId: 0, 'a199697', toolType: 1
06-03 10:55:49.751  3255  3789 D PowerManagerService: [api] userActivityFromNative : 10 (event: 2 flags: 0) eventTime = 353453443
06-03 10:55:49.753 20634 20634 D ViewRootImpl@a527994[MainActivity]: ViewPostIme pointer 0
06-03 10:55:49.849  2812  2910 I display : [PrimaryDisplay] [DYNAMIC_RECOMP] first frame after HWC_2_GLES
06-03 10:55:49.849  2812  2910 I display : [PrimaryDisplay] [DYNAMIC_RECOMP] GLES_2_HWC by high FPS(5)
06-03 10:55:49.916  3255  3790 D InputReader: Input event(11): value=0 when=353453.610190
06-03 10:55:49.916  3255  3790 I InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=353453.610190
06-03 10:55:49.916  3255  3789 I clp-JNI : Add pinfo) a199697 com.altairble/com.altairble.MainActivity (server) : AppWindowToken{50c5963 token=Token{2efcb92 ActivityRecord{d73fd1d u0 com.altairble/.MainActivity t550}}} : 20634 : 10161 : 1
06-03 10:55:49.916  3255  3789 I clp-JNI : Add pinfo) 1
06-03 10:55:49.916  3255  3789 I InputDispatcher: Delivering touch to (20634): action: 0x1, displayId: 0, 'a199697', toolType: 1
06-03 10:55:49.917 20634 20634 D ViewRootImpl@a527994[MainActivity]: ViewPostIme pointer 1
06-03 10:55:49.966 20634 20684 I ReactNativeJS: Logging Data
06-03 10:55:50.081 20634 21218 D NetworkManagementSocketTagger: tagSocket(87) with statsTag=0xffffffff, statsUid=-1
06-03 10:55:50.393  2826 15926 V APM_AudioPolicyManager: getAudioPolicyConfig: audioParam;outDevice
06-03 10:55:50.393  2826 15926 V APM_AudioPolicyManager: getNewOutputDevice() selected device 0
06-03 10:55:50.393  2826 15926 V APM_AudioPolicyManager: ### curdevice : 2
06-03 10:55:50.763  2812 19900 I display : [PrimaryDisplay] [DYNAMIC_RECOMP] HWC_2_GLES by low FPS(0)
06-03 10:55:51.348 20634 21220 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 10:55:51.503  2961  3050 D EnterpriseController: netId is 0
06-03 10:55:51.503  2961  3050 E Netd    : getNetworkForDns: getNetId from enterpriseCtrl is netid 0
06-03 10:55:51.503  2961  3050 D DnsProxyListener: DNSDBG::dns addrinfo af 0
06-03 10:55:51.507 20634 20654 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 10:55:51.892  5546  5546 D io_stats: !@ 179,0 r 170973 7601158 w 760974 17728608 d 57329 34264756 f 243335 243251 iot 565570 662473 th 51200 0 0 pt 0 inp 0 0 353455.584
06-03 10:55:52.061  2828  2828 I SurfaceFlinger: SFWD update time=353455756155265
06-03 10:55:52.126 20634 21222 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 10:55:52.863  5990  6142 D SDHMS:s : SIOP:: AP:199(246,40) BAT:231(231,0) CHG:0(0,0) USB:0(0,0) ATC:0(0,-999) 
06-03 10:55:53.368 20634 21223 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 10:55:53.522  2961  3050 D EnterpriseController: netId is 0
06-03 10:55:53.522  2961  3050 E Netd    : getNetworkForDns: getNetId from enterpriseCtrl is netid 0
06-03 10:55:53.522  2961  3050 D DnsProxyListener: DNSDBG::dns addrinfo af 0
06-03 10:55:53.527 20634 20654 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 10:55:54.178 20634 21225 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 10:55:54.453  3255  3752 D SensorService: [SO] -1.403 1.561 9.558
06-03 10:55:55.025  3255  5354 D BatteryService: Sending ACTION_BATTERY_CHANGED. scale:100, info:{.chargerAcOnline = false, .chargerUsbOnline = true, .chargerWirelessOnline = false, .maxChargingCurrent = 0, .maxChargingVoltage = 0, .batteryStatus = CHARGING, .batteryHealth = GOOD, .batteryPresent = true, .batteryLevel = 96, .batteryVoltage = 4290, .batteryTemperature = 231, .batteryCurrent = 0, .batteryCycleCount = 0, .batteryFullCharge = 3000000, .batteryChargeCounter = 2895000, .batteryTechnology = Li-ion}
06-03 10:55:55.025  3255  5354 D BatteryService: online:4, current avg:377, charge type:1, POGO powered:false, power sharing:false, high voltage charger:false, charger_type:0, capacity:280000, batterySWSelfDischarging:false, misc_event:0, current_event:0, current_now:405
06-03 10:55:55.028  3255  3255 I MotionRecognitionService: Plugged
06-03 10:55:55.028  3255  3255 D MotionRecognitionService:   mCableConnection= 1
06-03 10:55:55.028  3255  3255 D MotionRecognitionService: setPowerConnected : mGripEnabled = false
06-03 10:55:55.028  3255  3255 D SamsungPhoneWindowManager: ACTION_BATTERY_CHANGED - Level :: 96, battStatus :: 2
06-03 10:55:55.029  3255  3255 D UsbDeviceManager: received ACTION_BATTERY_CHANGED
06-03 10:55:55.030  3255  3280 D UsbDeviceManager: handleMessage -> MSG_UPDATE_CHARGING_STATE = 1
[... cut ...]
06-03 10:56:19.757  2961  3050 D DnsProxyListener: DNSDBG::dns addrinfo af 0
06-03 10:56:19.762 20634 20654 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 10:56:19.836  5250  5516 D NetworkController.MobileSignalController(0/-2): onSignalStrengthsChanged signalStrength=SignalStrength: 8 99 -8 -200 -8 -200 -1 99 2147483647 2147483647 2147483647 2147483647 0 2147483647 8 24 -96 0x3 P gsm|lte use_rsrp_and_rssnr_for_lte_level  [-128, -118, -108, -98] [-115, -105, -95, -85] level=3
```

dariuszseweryn, 2019-06-03:
You did not enabled logs. Enable them

psambit9791, 2019-06-03:
Do you mean I need to set the Log Level to Verbose?

dariuszseweryn, 2019-06-03:
Yes

psambit9791, 2019-06-03:
Adding the Logs with log level set to Verbose:

```
06-03 12:50:35.457  3255  3789 I InputDispatcher: Delivering touch to (937): action: 0x1, displayId: 0, '5233e72', toolType: 1
06-03 12:50:35.462   937   937 D ViewRootImpl@b693185[MainActivity]: ViewPostIme pointer 1
06-03 12:50:35.505   937  1530 I ReactNativeJS: Logging Data
06-03 12:50:36.141   937  1641 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:36.508  3255  3790 D InputReader: Input event(11): value=1 when=360340.202201
06-03 12:50:36.508  3255  3790 I InputReader: Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.2969 ] when=360340.202201
06-03 12:50:36.508  3255  3789 I InputDispatcher: Delivering touch to (5250): action: 0x4, displayId: 0, '8445d9 ', toolType: 1
06-03 12:50:36.508  3255  3789 I InputDispatcher: Delivering touch to (5250): action: 0x4, displayId: 0, '42ee231', toolType: 1
06-03 12:50:36.509  3255  3789 I clp-JNI : Add pinfo) 0
06-03 12:50:36.509  3255  3789 I InputDispatcher: Delivering touch to (937): action: 0x0, displayId: 0, '5233e72', toolType: 1
06-03 12:50:36.511   937   937 D ViewRootImpl@b693185[MainActivity]: ViewPostIme pointer 0
06-03 12:50:36.549  5990  6142 D SDHMS:s : SIOP:: AP:240(282,40) BAT:277(277,0) CHG:0(0,0) USB:0(0,0) ATC:0(0,-999) 
06-03 12:50:36.604  2961  3050 D EnterpriseController: netId is 0
06-03 12:50:36.604  2961  3050 E Netd    : getNetworkForDns: getNetId from enterpriseCtrl is netid 0
06-03 12:50:36.604  2961  3050 D DnsProxyListener: DNSDBG::dns addrinfo af 0
06-03 12:50:36.606   937   978 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:36.608  3255  3790 D InputReader: Input event(11): value=0 when=360340.302373
06-03 12:50:36.608  3255  3790 I InputReader: Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=360340.302373
06-03 12:50:36.608  3255  3789 I clp-JNI : Add pinfo) 5233e72 com.altairble/com.altairble.MainActivity (server) : AppWindowToken{6cf0229 token=Token{c023b0 ActivityRecord{3384af3 u0 com.altairble/.MainActivity t554}}} : 937 : 10161 : 1
06-03 12:50:36.608  3255  3789 I clp-JNI : Add pinfo) 1
06-03 12:50:36.608  3255  3789 I InputDispatcher: Delivering touch to (937): action: 0x1, displayId: 0, '5233e72', toolType: 1
06-03 12:50:36.609   937   937 D ViewRootImpl@b693185[MainActivity]: ViewPostIme pointer 1
06-03 12:50:36.639   937  1530 I ReactNativeJS: Delete Data
06-03 12:50:36.645   937  1531 D RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(262620518)
06-03 12:50:36.647   937  1552 D RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(262620518)
06-03 12:50:36.651   937  1548 D BluetoothGatt: configureMTU() - device: 08:6B:D7:41:BA:79 mtu: 46
06-03 12:50:36.735   937   996 D BluetoothGatt: onConfigureMTU() - Device=08:6B:D7:41:BA:79 mtu=46 status=0
06-03 12:50:36.737   937   996 D RxBle#BluetoothGatt$1: onMtuChanged mtu=46 status=0
06-03 12:50:36.745   937  1552 D RxBle#ConnectionOperationQueue: FINISHED MtuRequestOperation(262620518) in 97 ms
06-03 12:50:36.829   937  1530 I ReactNativeJS: 'Notification activated', '08:6B:D7:41:BA:79'
06-03 12:50:36.834   937  1531 D RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(234123957)
06-03 12:50:36.835   937  1552 D RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(234123957)
06-03 12:50:36.865   937   971 D NetworkManagementSocketTagger: tagSocket(57) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:37.217   937  1643 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:37.696  2826 28376 V APM_AudioPolicyManager: getAudioPolicyConfig: audioParam;outDevice
06-03 12:50:37.696  2826 28376 V APM_AudioPolicyManager: getNewOutputDevice() selected device 0
06-03 12:50:37.696  2826 28376 V APM_AudioPolicyManager: ### curdevice : 2
06-03 12:50:38.161   937  1644 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:38.617  2961  3050 D EnterpriseController: netId is 0
06-03 12:50:38.617  2961  3050 E Netd    : getNetworkForDns: getNetId from enterpriseCtrl is netid 0
06-03 12:50:38.617  2961  3050 D DnsProxyListener: DNSDBG::dns addrinfo af 0
06-03 12:50:38.621   937   978 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:39.271   937  1646 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:40.180   937  1647 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:40.634  2961  3050 D EnterpriseController: netId is 0
06-03 12:50:40.634  2961  3050 E Netd    : getNetworkForDns: getNetId from enterpriseCtrl is netid 0
06-03 12:50:40.634  2961  3050 D DnsProxyListener: DNSDBG::dns addrinfo af 0
06-03 12:50:40.638   937   978 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:41.324   937  1649 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:41.877   937   971 D NetworkManagementSocketTagger: tagSocket(56) with statsTag=0xffffffff, statsUid=-1
06-03 12:50:41.903  5211  5832 W bt_stack: [WARNING:bta_gattc_act.cc(1069)] bta_gattc_conn_cback: cif=3 connected=0 conn_id=0x0003 reason=0x0008
06-03 12:50:41.904  5211  5832 W bt_stack: [WARNING:bta_gattc_act.cc(1069)] bta_gattc_conn_cback: cif=4 connected=0 conn_id=0x0004 reason=0x0008
[... cut ...]
06-03 12:50:44.078   937  1530 W ReactNativeJS: invokeCallbackAndReturnFlushedQueue@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2683:33
06-03 12:50:44.078   937  1530 W ReactNativeJS: invokeCallbackAndReturnFlushedQueue@[native code]
06-03 12:50:44.102  5250  5516 D NetworkController.MobileSignalController(0/-2): onSignalStrengthsChanged signalStrength=SignalStrength: 8 99 -8 -200 -8 -200 -1 99 2147483647 2147483647 2147483647 2147483647 0 2147483647 8 23 -97 0x3 P gsm|lte use_rsrp_and_rssnr_for_lte_level  [-128, -118, -108, -98] [-115, -105, -95, -85] level=3
```

dariuszseweryn, 2019-06-03:
I see what is happening. You can filter the logs for `RxBle#` and you will see all relevant logs. Thing is this is not the same error that you have stated before. In the above log the problem is:
```
06-03 12:50:44.078   937  1530 W ReactNativeJS: BleError: Characteristic 04b30006-6f01-4987-bf0e-f8aecc7b42a8 write failed for device 08:6B:D7:41:BA:79 and service 04b30000-6f01-4987-bf0e-f8aecc7b42a8
```
Where previously you stated:
```
06-01 21:12:07.653 15001 15052 I ReactNativeJS: {"message":"Characteristic 04b30006-6f01-4987-bf0e-f8aecc7b42a8 notify change failed for device ? and service 04b30000-6f01-4987-bf0e-f8aecc7b42a8","errorCode":403,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":"Cannot write client characteristic config descriptor (code 3) with characteristic UUID 04b30006-6f01-4987-bf0e-f8aecc7b42a8","name":"BleError","line":86743,"column":107,"sourceURL":"http://localhost:8081/index.delta?platform=android&dev=true&minify=false"}
```
Anyway from the above logs it seems that for some reason your peripheral has a timeout — possibly it has an issue in the firmware. I do not know what that is — you should ask the firmware developer

psambit9791, 2019-06-03:
When I ran the application another time, this is what came in the logs. The previous one had write faiure, but now, it turns up as a notify change failed:

```
06-03 12:16:22.799 29615 29738 D RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(159197218) in 4901 ms
06-03 12:16:22.801  5211  5343 D BluetoothUtils: getBtEnabledContainers(): btContainers = []
06-03 12:16:22.801  5211  5343 E BluetoothRemoteDevices: Remote class is UNCATEGORIZED
06-03 12:16:22.808 29615 29739 D BluetoothGatt: setCharacteristicNotification() - uuid: 04b30006-6f01-4987-bf0e-f8aecc7b42a8 enable: false
06-03 12:16:22.809  5211  9339 E BtGatt.GattService: registerForNotification() - No connection for 41:B...
06-03 12:16:22.811 29615 29628 D RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=8
06-03 12:16:22.813 29615 29739 D RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(164002055)
06-03 12:16:22.813 29615 29628 I RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (08:6B:D7:41:BA:79)
06-03 12:16:22.815 29615 29738 D RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(164002055)
06-03 12:16:22.827  5250  5516 V BluetoothEventManager: onReceive :: android.bluetooth.device.action.NAME_CHANGED
06-03 12:16:22.832  3489  3489 I WorkerManager: New handler created for worker [service] (manager: d90b2c1, handler: f591a66)
06-03 12:16:22.832  3489  3489 I WorkerHandler: New state for worker [service]: e (handler: f591a66)
06-03 12:16:22.833 29615 29739 D RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(96556330)
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: Queue's awaitRelease() has been interrupted abruptly while it wasn't released by the release() method.
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: java.lang.InterruptedException
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: 	at java.lang.Object.wait(Native Method)
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: 	at com.polidea.rxandroidble.internal.serialization.QueueSemaphore.awaitRelease(QueueSemaphore.java:15)
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: 	at com.polidea.rxandroidble.internal.serialization.ConnectionOperationQueueImpl$1.run(ConnectionOperationQueueImpl.java:73)
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:459)
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
06-03 12:16:22.835 29615 29738 W RxBle#ConnectionOperationQueue: 	at java.lang.Thread.run(Thread.java:764)
06-03 12:16:22.840 29615 29699 D RxBle#ClientOperationQueue: STARTED  DisconnectOperation(96556330)
06-03 12:16:22.843 29615 29738 D RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(164002055) in 29 ms
06-03 12:16:22.844 29615 29738 D RxBle#Executors$RunnableAdapter: Terminated.
06-03 12:16:22.860 29615 29699 D RxBle#ClientOperationQueue: FINISHED DisconnectOperation(96556330) in 20 ms
06-03 12:16:22.884 29615 29666 I ReactNativeJS: {"message":"Characteristic 04b30006-6f01-4987-bf0e-f8aecc7b42a8 notify change failed for device ? and service 04b30000-6f01-4987-bf0e-f8aecc7b42a8","errorCode":403,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":"Cannot write client characteristic config descriptor (code 3) with characteristic UUID 04b30006-6f01-4987-bf0e-f8aecc7b42a8","name":"BleError","line":86743,"column":107,"sourceURL":"http://localhost:8081/index.delta?platform=android&dev=true&minify=false"}
```

Apparently, the error seems to alternate. If I use the app for the first time, the write fails. If I reconnect to the device and try again, then it is a notify change fail and it continues like that.

dariuszseweryn, 2019-06-03:
The notification fails, that is true, but it is all because the peripheral times out. If you would monitor `Device.onDisconnect(listener)` you should be able to get there the disconnection reason. Anyway — the library cannot mitigate the problem because it is the peripheral who timesout.

psambit9791, 2019-06-03:
Alright Thanks a lot. Will take a look from the firmware.

closed 2019-06-03 by dariuszseweryn