dotintent/react-native-ble-plx #481: All devices scanned have name equals to null

TailNest, 2019-06-05:
## Expected Behavior

Scanning devices should return a name in order to understand which device is which.

## Current Behavior

Scanning devices return objects with `name` and `localName` equals `null`:

```
Device {
    id : "2C:F0:EE:19:27:CF"
    isConnectable : null 
    localName : null
    manufacturerData : "TAAQBQsUluNJ"
    mtu : 23
    name : null
    overflowServiceUUIDs : null
    rssi : -62 
    serviceData : null 
    serviceUUIDs : null 
    solicitedServiceUUIDs : null 
    txPowerLevel : null
    _manager : BleManager
    __proto__ : Object
}
```

### Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions. Both JS and platform logs can be enabled via [setLogLevel](https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel) function call. 

* Library version: 1.0.3
* Platform: currently tested only in Android
* Contents of the `package.json` file:
```
  "dependencies": {
    "react": "16.8.3",
    "react-native": "0.59.8",
    "react-native-ble-plx": "^1.0.3",
    "react-native-gesture-handler": "^1.3.0",
    "react-navigation": "^3.11.0"
  },
  "devDependencies": {
    "@babel/core": "^7.4.5",
    "@babel/runtime": "^7.4.5",
    "babel-jest": "^24.8.0",
    "jest": "^24.8.0",
    "metro-react-native-babel-preset": "^0.54.1",
    "react-test-renderer": "16.8.3"
  },
```

* Formatted code sample or link to a repository:

```
import React from "react";
import { PermissionsAndroid, View, Text, Button } from "react-native";
import { BleManager } from 'react-native-ble-plx';    

const ProfileScreen = props => {
  async function requestCameraPermission() {
    try {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
        {
          title: 'Cool Photo App Camera Permission',
          message: 'Cool Photo App needs access to your camera',
          buttonNeutral: 'Ask Me Later',
          buttonNegative: 'Cancel',
          buttonPositive: 'OK',
        },
      );
      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        console.log('You can use the camera');
        return true;
      } else {
        console.log('Camera permission denied');
        return false;
      }
    } catch (err) {
      console.warn(err);
      return false;
    }
  }

  scanAndConnect = () => {
    const permission = requestCameraPermission();

    if (permission) {
      const bluetoothInstance = new BleManager();

      bluetoothInstance.startDeviceScan(null, null, (error, device) => {
        if (error) {
          console.log(error);
          return
        } else {
          console.log(device);
        }
      });
    } else {
      console.log("permission not granted")
    }
  };

  return (
    <View style={{ flex: 1, alignItems: "center", justifyContent: "center" }}>
      <Text>Profile</Text>
      <Button 
        title="Start scanning"
        onPress={scanAndConnect}/>
      </View>
  )
}
```

dariuszseweryn, 2019-06-05:
Devices do not have to advertise name as per Core Bluetooth Specification

dariuszseweryn, 2019-06-05:
For identifying a device use it's `id`

closed 2019-06-05 by dariuszseweryn

RWOverdijk, 2019-06-23:
@dariuszseweryn Sorry but that's a bit confusing, could you confirm my understanding is correct?

You have to use the id to identify the device but it's different on iOS and android, so we should then store two ids and compare them based on the OS?

dariuszseweryn, 2019-06-24:
You can compare ids between Android devices if peripheral uses a static MAC address*. iOS ids are not comparable between devices i.e. the same peripheral will have different ids on two iOS devices.

\* core bluetooth specification can teach you more about MAC addresses in Bluetooth.

RWOverdijk, 2019-06-24:
>  i.e. the same peripheral will have different ids on two iOS devices

If this is true I'm really screwed because my entire product requires this. How can I identify a beacon if the ID is different on every device? Are you sure about this?

If this is true, do I have to use the device name instead?

dariuszseweryn, 2019-06-24:
Name is usually quite wobbly. It could be major, minor, region, uuid or other unique things that beacons usually advertise. Go figure what is the best idea for your case

RWOverdijk, 2019-06-26:
Why is name wobbly? I get localName on iOS and Android. It blows my mind that there's no unique identifier that's consistent 😄

So, I guess I'll have to learn about major and minor then? Is there a guide on how to identify the beacons consistently? All I need is a unique identifier and the rssi, nothing more.

dariuszseweryn, 2019-06-26:
It is not easy to filter by name simply because it is a string and it is hard to make it readable and unique. Unique ID is bad because of privacy and ability to locate the user. Good luck on learning how to solve your use case

RWOverdijk, 2019-06-26:
Thanks. And if you ever find out how beacons work do share! I really want to solve this. If this doesn't work I see no use for beacons other than spam devices.

sudomann, 2020-02-19:
I don't understand how this library reports device `name` and `localName` as `null`. I simply logged all detected devices, and every single one, `null`. Using my smartphone to see all bluetooth devices in the area reveals them with human readable names. Therefore, they are reporting them. I instantiated a manager and scanned for devices. Is there a step i missed?

tyrone2kx, 2020-07-21:
@sudomann Did you find your way around this? I have the same exact issue. It also happens that when there's no available bluetooth device, my bluetooth scan still prints a device id.

sudomann, 2020-07-21:
@tyrone2kx oddly enough, it started displaying showing the names soon after. I also noticed sometimes the scan results had duplicates. One might show the name, while the other might not.

It might help to log all the scan results and closely inspect.

SatyaGanesh399, 2023-02-09:
does anyone figuredout how to handle the problem. because i could nt find the problem in the whole internet.. all the names and localnames of the devices are returning null. Please can one tell me how to display the names of the devices

aliwaqar981, 2024-06-21:
Does anyone found a solution for this? All the names and localnames are null for all devices.

aliwaqar981, 2024-06-24:
@SatyaGanesh399 have you found solution for this?

SatyaGanesh399, 2024-06-25:
@aliwaqar981 nope. i'm not woring on react native anymore