dotintent/react-native-ble-plx #482: ScanCallbackType not working as expected

yoppoy, 2019-06-07:
Hi,

As my app needs to scan about 50 ble devices around it, optimisation is important on my end. I wanted to change scanCallbackType to FirstMatch. I did so to notice no apparent change in behavior, the same amount of callbacks were called.

I went ahead in the native android code to change in the BleModule.java file the default callback type from (line 375) :
```int callbackType = CALLBACK_TYPE_ALL_MATCHES;```
to :
```int callbackType = CALLBACK_TYPE_FIRST_MATCH;```
After doing that, every time I start a scan I get this error : 
```BleError: Cannot start scanning operation```

Any ideas how I make first match work ? I am using the latest version of react-native-ble-plx.

Thanks in advance

dariuszseweryn, 2019-06-07:
Have you checked the [wiki FAQ](https://github.com/Polidea/react-native-ble-plx/wiki/=-FAQ:-I-got-a-BleError.-Now-what%3F)?

yoppoy, 2019-06-07:
Thanks for your quick response. I hadn't had a read at it. I was just printing the BleError without stringifying it, turns out the feature isn't supported on my device sadly. Have a great day and thank you again for your hard work

closed 2019-06-07 by yoppoy

AjayParambath, 2022-06-06:
> 
> ```
> this.manager.startDeviceScan(null, { callbackType: ScanCallbackType.FirstMatch}, (error, device) => {
>       if (error) {
>         this.props.error(error);
>         return
>       }
>       this.addScannedDevice(device);
>     });
> ```
> try this