dotintent/react-native-ble-plx #483: Parsing Manufacturer Specific Data when using Scan Response Event Type

longchiwen, 2019-06-11:
It looks like **Android** does not capture **Manufacturer Specific Data** correctly, when using **Scan Response Event Type** together with **Connectable Undirected Advertising Event Type**. Bluetooth Specification v4.2 suggests, that it is possible to have Manufacturer Specific Data in AD and in Scan Response package. On IOS this is correctly handled by concatenating data between the two, while Android only captures Manufacturer Specific Data from the last Advertising package (Scan Response).

## Expected Behavior

Capturing ManufacturerData should be consistent across IOS and Android devices. If Scan Response package with Manufacturer Specific Data is received, that data should be appended to Manufacturer Specific Data from initial AD.

## Current Behavior

Android is skipping Manufacturer Specific Data from initial Advertising Package, if device sent Scan Response data after that. 

Let's say device sends 2 advertising packages:

#### 1. Advertising package - Event Type: Connectable Undirected Advertising (0x00) 
Broken down Manufacturer Data part of Advertising package and image of the captured package
```
┌── Length
|   ┌── Type (Manufacturer Specific)
|   |   ┌── Company ID (Nordic Semiconductor ASA)
|   |   |     ┌── Data (15 bytes)
|   |   |     |
┌──┐┌──┐┌────┐┌──────────────────────────────┐
 12  ff  5900  000000000000000000000000000000
```
[image]

#### 2. Advertising package - Event Type: Scan Response (0x04)
Broken down Manufacturer Data part of Advertising package and image of the captured package
```
┌── Length
|   ┌── Type (Manufacturer Specific)
|   |   ┌── Company ID (Nordic Semiconductor ASA)
|   |   |     ┌── Data (27 bytes)
|   |   |     |
┌──┐┌──┐┌────┐┌──────────────────────────────────────────────────────┐
 1e  ff  5900  001d00290018504b35433836200000000001000000000000000000
```
[image]

Currently, Android skips the data in the first package, while IOS returns both

#### Android
This is the Manufacturer data returned on Android (base64 decoded)
```
┌── Company ID (Nordic Semiconductor ASA)
|     ┌── Data (27 bytes)
|     |
┌────┐┌──────────────────────────────────────────────────────┐
 5900  001d00290018504b35433836200000000001000000000000000000
```

#### IOS
This is the Manufacturer data returned on IOS (base64 decoded) 
```
┌── Company ID (Nordic Semiconductor ASA)
|     ┌── Data (15 bytes)
|     |                               ┌── Data (27 bytes)
|     |                               |
┌────┐┌──────────────────────────────┐┌──────────────────────────────────────────────────────┐
 5900  000000000000000000000000000000  001d00290018504b35433836200000000001000000000000000000
```

### Steps to Reproduce
You can reproduce this on devices that send Scan Response packages with Manufacturer Specific Data.
1. Scan for devices
2. Print out `manufacturerData` property on Android
3. Print out `manufacturerData` property on IOS (they should match)

### Context

* Library version: 1.0.3 (same problem on current master branch - commit `72f38d078d4ff61ca914c4a418d829ac3bb505c8`)
* Platform: Android 9 (API 28)

closed 2019-08-17 by stale[bot]

blacksheep--, 2019-10-11:
I encoutered the very same behaviour. What should be done here?

rubiomiguel06, 2020-06-05:
I'm facing the very same behaviour though only with Android versions lower than the 9.0. With Android 9.0 I'm getting what it's expeceted to be (Advertisement + scanResponse). But for the previous Android versions (I''ve tested on Android 8.0 - API 26 and Android 6.0 - API 23) the ManufacturerSpecificData from the advertisement is not present.

In fact, while debugging I am able to see that the Raw Data of the scan result has the complete 'packet' (Advertisement + scanResponse), but the problem appears when using Android API to parse that packet. If I get the ManufacturerSpecificData, I only get the Manufacturer Specific Data from the scanResponse.

If I do the same experiment on Android 9.0 - API 28, the getter of ManufacturerSpecificData returns the concatenation of the manuf. specific data of the advertisement and the scanResponse (as expected)

I don't think this is a flaw in the bluetooth peripheral's firmware as @blacksheep-- stated because I'm having the same problem on several devices.