dotintent/react-native-ble-plx #484: Using BLE in background on Android

keyz23, 2019-06-12:
Hi everyone, I use this library on my Android App, but I need to use also BLE in background! It's possible to do on Android Device with this module?
Thank you

closed 2019-08-18 by stale[bot]

njdullea, 2019-08-22:
Hey @keyz23 did you ever get an answer?

curest0x1021, 2019-10-13:
I need to get the BLE signals even though the app is in the background on both of iOS and Android devices.
@keyz23 Did you solve the background issue? If you already solved this problem please let me know the solution.

Thanks.

dariuszseweryn, 2020-03-12:
On Android one needs to implement a Foreground Service to keep the Application process running. You can find more info on that using Google.

ezranbayantemur, 2020-03-12:
> On Android one needs to implement a Foreground Service to keep the Application process running. You can find more info on that using Google.

Thank you for your quick response.
I haven't try but does [react-native-background-task](https://github.com/jamesisaac/react-native-background-task) package handles foreground service implementation?

dariuszseweryn, 2020-03-12:
> I haven't try but does react-native-background-task package handles foreground service implementation?

I am not familiar with this library

jordan-tpz, 2021-01-06:
> On Android one needs to implement a Foreground Service to keep the Application process running. You can find more info on that using Google.

@dariuszseweryn If I want to keep the connection with my BLE device in background on Android, I understood I have to create a specific task (using foreground service). 

My question is : do you think I should use the `connectToDevice()` function in this task when I connect my BLE device with my app ? (and I could remove this task when I would like to disconnect my BLE device)
But, another solution : maybe quite simply I should create the BleManager instance within this task ?

dariuszseweryn, 2021-01-06:
It all depends on how you want to architecture your app. This topic is very wide and subjective — personally I tend to have a single object managing the connection.