dotintent/react-native-ble-plx #485: Return if location services are disabled on Android from state() call

CatapultJesse, 2019-06-16:
## Current Behavior
When checking for state (https://github.com/Polidea/react-native-ble-plx/wiki/Bluetooth-Adapter-State) it doesn't cover the case if Android Location services are turned off

## New behavior
The underlying library seems to be able to determine this: https://github.com/Polidea/RxAndroidBle#observing-client-state

### Acceptance criteria
Return the location services disabled as an additional option from the state() call

### Breaking changes
I believe atm if Location services are off it will return the adapter is 'PoweredOn', so some of these will swap to returning 'LocationServicesDisabled' (or whatever constant is decided)

CatapultJesse, 2019-08-15:
Still a requirement and should be reasonably straight forward to add

closed 2019-10-21 by stale[bot]

laithmm, 2021-09-03:
Did this get added? It seems important since BLE will not work if location is not turned on