dotintent/react-native-ble-plx #486: Returning indication values from monitorCharacteristicForDevice

psambit9791, 2019-06-19:
Is there any way to continuously return values as they are updated in the monitorCharacteristicForDevice.

I am writing a RN wrapper for a device using ble-plx so I cannot use redux to share states between files. What I am struggling with is that the notification/indication that reaches the monitorCharacteristicForDevice method cannot be returned to any other method regularly.

So, this is my code on the wrapper end:

```
activateNotificationForBattery = async (BLEMgr, deviceID) => {
        await BLEMgr.readCharacteristicForDevice(deviceID, SERVICE_UUID, CHARACTERISTIC_UUID.BATTERY.read).then((value) => {
            this.battery_level.Voltage = getBatteryVoltage(value.value);
            this.battery_level.Percentage = convertVoltageToPercentage(this.battery_level.Voltage);
            console.log(this.battery_level);
        });
        this.battery_notif = await BLEMgr.monitorCharacteristicForDevice(deviceID, SERVICE_UUID, CHARACTERISTIC_UUID.BATTERY.read, async (error, characteristic) => {
            if (error) {
                console.log(JSON.stringify(error));
                return
            };
            this.battery_level.Voltage = await getBatteryVoltage(characteristic.value);
            this.battery_level.Percentage = await convertVoltageToPercentage(this.battery_level.Voltage);
            console.log(this.battery_level);
            return new Promise(resolve => {
                resolve(this.battery_level);
            })
        });
    }
```

And on the app end:

```
this.battObj = new AltairBLEWrapper();
                    await this.battObj.activateNotificationForBattery(this.manager, this.device.id);
                    console.log("Home", this.battObj.battery_level);
                    this.props.dispatch(battery_level({PERCENTAGE: this.battObj.battery_level.Percentage}));
```
I just get one update because of the readCharacteristicForDevice().
At no point does the return actually send the updated values back. Is there any way this can be achieved such that the values are sent back as soon as an update is received?

psambit9791, 2019-06-19:
Worked out the issue using RxJS Observables within the wrapper.
Used setInterval() for the notification timeframe within the Observable object to get persistent notifications and then subscribed to it from the app.

closed 2019-06-19 by psambit9791

fitbuff, 2020-08-19:
hey @psambit9791 ,
are you applying the `setInterval()` to `monitorCharacteristicForDevice`?

so something like?
```
setInterval(async () =>
      await BLEMgr.monitorCharacteristicForDevice(
          deviceID, SERVICE_UUID, CHARACTERISTIC_UUID.BATTERY.read, async (error, characteristic) => {
      })
)
```

psambit9791, 2020-08-19:
@fitbuff

Since the BLEMgr.monitorCharacteristicForDevice can already be setup in a variable, the "characteristic" is the variable that needs to be checked within setInterval().
Looping through BLEMgr.monitorCharacteristicForDevice multiple times will only renew the monitoring service.

fitbuff, 2020-08-20:
thanks @psambit9791 
im new to Bluetooth so bear with but if i understand correctly your suggesting id do:

```
cont value = await BLEMgr.monitorCharacteristicForDevice(....);
setInterval(() => value, 1000);
```

but if i `console.log(value)` in `setInterval` it'll show `{"remove": [Function remove]}`

what am i missing here?