dotintent/react-native-ble-plx #487: BleError: Characteristic ... read failed for device

ExoMemphiz, 2019-06-19:
- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

## Expected Behavior

Reading characteristic from Glucose Monitoring Device delivers readable value.

## Current Behavior

I try reading the characteristic, and I get the following error:

```BleError: Characteristic 00002a51-0000-1000-8000-00805f9b34fb read failed for device CC:78:AB:A3:A8:97 and service 00001808-0000-1000-8000-00805f9b34fb```

### Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Connect to the device normally.
2. 
```const GLUCOSE = {
		char: "00002a51-0000-1000-8000-00805f9b34fb",
		service: "00001808-0000-1000-8000-00805f9b34fb",
	};
	const readPromise = device.readCharacteristicForService(
		GLUCOSE.service,
		GLUCOSE.char,
	);```
3. ```BleError: Characteristic 00002a51-0000-1000-8000-00805f9b34fb read failed for device CC:78:AB:A3:A8:97 and service 00001808-0000-1000-8000-00805f9b34fb```

### Context

Not very useable React Native log:

```06-19 17:14:30.667 32260 32679 W ReactNativeJS: BleError: Characteristic 00002a51-0000-1000-8000-00805f9b34fb read failed for device CC:78:AB:A3:A8:97 and service 00001808-0000-1000-8000-00805f9b34fb
06-19 17:14:30.667 32260 32679 W ReactNativeJS: construct@[native code]
06-19 17:14:30.667 32260 32679 W ReactNativeJS: _construct@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:94079:28
06-19 17:14:30.667 32260 32679 W ReactNativeJS: Wrapper@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:94023:25
06-19 17:14:30.667 32260 32679 W ReactNativeJS: BleError@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:93851:106
06-19 17:14:30.667 32260 32679 W ReactNativeJS: parseBleError@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:93889:30
06-19 17:14:30.667 32260 32679 W ReactNativeJS: _callPromise$@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:94237:51
06-19 17:14:30.667 32260 32679 W ReactNativeJS: tryCatch@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:23068:23
06-19 17:14:30.667 32260 32679 W ReactNativeJS: invoke@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:23243:32
06-19 17:14:30.667 32260 32679 W ReactNativeJS: tryCatch@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:23068:23
06-19 17:14:30.667 32260 32679 W ReactNativeJS: invoke@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:23144:30
06-19 17:14:30.667 32260 32679 W ReactNativeJS: http://localhost:8081/index.delta?platform=android&dev=true&minify=false:23156:21
06-19 17:14:30.667 32260 32679 W ReactNativeJS: tryCallOne@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:24360:16
06-19 17:14:30.667 32260 32679 W ReactNativeJS: http://localhost:8081/index.delta?platform=android&dev=true&minify=false:24461:27
06-19 17:14:30.667 32260 32679 W ReactNativeJS: _callTimer@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:27743:17
06-19 17:14:30.667 32260 32679 W ReactNativeJS: _callImmediatesPass@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:27779:19
06-19 17:14:30.667 32260 32679 W ReactNativeJS: callImmediates@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:27998:33
06-19 17:14:30.667 32260 32679 W ReactNativeJS: callImmediates@[native code]
06-19 17:14:30.667 32260 32679 W ReactNativeJS: __callImmediates@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2548:35
06-19 17:14:30.667 32260 32679 W ReactNativeJS: http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2369:34
06-19 17:14:30.667 32260 32679 W ReactNativeJS: __guard@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2531:15
06-19 17:14:30.667 32260 32679 W ReactNativeJS: flushedQueue@http://localhost:8081/index.delta?platform=android&dev=true&minify=false:2368:21
06-19 17:14:30.667 32260 32679 W ReactNativeJS: flushedQueue@[native code]
06-19 17:14:30.667 32260 32679 W ReactNativeJS: invokeCallbackAndReturnFlushedQueue@[native code]```

* Library version: 1.0.3
* Platform: Android.
* Contents of the `package.json` file:
```"dependencies": {
		"@types/js-base64": "^2.3.1",
		"js-base64": "^2.5.1",
		"react": "16.8.3",
		"react-native": "0.59.9",
		"react-native-ble-plx": "^1.0.3"
	},
	"devDependencies": {
		"@babel/core": "^7.4.5",
		"@babel/runtime": "^7.4.5",
		"@types/jest": "^24.0.15",
		"@types/react": "^16.8.20",
		"@types/react-native": "^0.57.63",
		"@types/react-test-renderer": "^16.8.2",
		"babel-jest": "^24.8.0",
		"jest": "^24.8.0",
		"metro-react-native-babel-preset": "^0.54.1",
		"react-test-renderer": "16.8.3",
		"typescript": "^3.5.2"
	},```

ExoMemphiz, 2019-06-19:
And I have ensured that the service and characteristic exists on the connected device.

closed 2019-06-22 by ExoMemphiz

renatoramossilva, 2020-10-24:
Hi,

I have the same problem. Can you show how you fixed this error ?

Gitika09, 2022-01-31:
Hey,
Even i facing the same issue.
Did you guys got the fix and how did you fixed it??

Please help....

davronakil, 2022-02-05:
Same error here after cloning a 32GB card (overloaded) onto a brand new 64GB sd card and placing it in a RAK v2 hotspot miner. Pairing fine and failing to open diagnostics.

ruialves35, 2022-04-06:
I'm having the same issue. I discover all the characteristics and services, i get the service i want also as the caracteristic, i check that the characteristic has the readable flag to true but then i get this error. Even more, this just work perfect on some tablets but fails in another one.

harrisol22, 2023-01-27:
Same issues, any movement?

milobob, 2023-08-20:
I had the same issue. However, I was using a sideloaded device for local debugging. The local device settings were "JS Dev Mode... with __DEV__ = true".. and Fast Refresh Disabled. When I turned JS Dev Mode OFF, and set to Fast Refresh, the problems went away and it worked. I imagine that JS Dev Mode off and Fast Refresh on reduce the latency. For a Bluetooth connection (doing stuff every 30ms or so during a connection) - a few tens of milliseconds of latency can be deadly. Happy coding, I hope this message helps someone.

lwhallock, 2024-05-08:
I have the same issue